robust-rosin/robust #2: Create an appropriately named DockerHub organisation

ChrisTimperley, 2018-04-05:


wasowski, 2018-04-05:
This name (`rosin`) is fine.  A lot of coauthors are funding this work via ROSIN, so it is only fair to somehow acknowledge this in the name (IMHO).

ChrisTimperley, 2018-04-06:
I've created the `rosin` organisation. @wasowski, if you send me your DockerHub username, I'll make you the admin.

gavanderhoorn, 2018-04-06:
Sorry for commenting so late, but could we just use the name of the Github organisation?

ROSIN is the name of the EU project and I'd like to avoid confusion.

ChrisTimperley, 2018-04-06:
Sure. I've created a new organisation named `robustrosin` (it looks like dashes aren't allowed).

wasowski, 2018-04-09:
my account is wasowski (but please don't make me the only admin - I am a newbie with docker).

gavanderhoorn, 2018-04-09:
Are underscores allowed? This is me micro-managing -- so feel free to ignore -- but `rosbust_rosin` looks a bit nicer than `robustrosin`.

ChrisTimperley, 2018-04-16:
@gavanderhoorn nice suggestion, but it appears that hyphens and underscores aren't allowed.

ChrisTimperley, 2018-04-16:
https://hub.docker.com/u/robustrosin/dashboard/

gavanderhoorn, 2018-04-16:
I would say this is done then :)