robust-rosin/robust #3: Possible missing dependencies for eed104d

ChrisTimperley, 2018-04-05:
```
==> Processing catkin package: 'gazebo_ros'
==> Creating build directory: 'build_isolated/gazebo_ros'
==> Building with env: '/opt/ros/env.sh'
==> cmake /ros_ws/src/gazebo_ros -DCATKIN_DEVEL_PREFIX=/ros_ws/devel_isolated/gazebo_ros -DCMAKE_INSTALL_PREFIX=/opt/ros -DCMAKE_BUILD_TYPE=Release -G Unix Makefiles in '/ros_ws/build_isolated/gazebo_ros'
-- The C compiler identification is GNU 5.4.0
-- The CXX compiler identification is GNU 5.4.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Using CATKIN_DEVEL_PREFIX: /ros_ws/devel_isolated/gazebo_ros
-- Using CMAKE_PREFIX_PATH: /opt/ros
-- This workspace overlays: /opt/ros
-- Found PythonInterp: /usr/bin/python (found version "2.7.12") 
-- Using PYTHON_EXECUTABLE: /usr/bin/python
-- Using Debian Python package layout
-- Using empy: /usr/bin/empy
-- Using CATKIN_ENABLE_TESTING: ON
-- Call enable_testing()
-- Using CATKIN_TEST_RESULTS_DIR: /ros_ws/build_isolated/gazebo_ros/test_results
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Found gtest sources under '/usr/src/gtest': gtests will be built
-- Using Python nosetests: /usr/bin/nosetests-2.7
-- catkin 0.6.18
-- Using these message generators: gencpp;genlisp;genpy
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29.1") 
-- Checking for module 'libxml-2.0'
--   No package 'libxml-2.0' found
CMake Error at CMakeLists.txt:26 (find_package):
  By not providing "Findgazebo.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration f
Command failed, exiting.

ile provided by "gazebo", but
  CMake did not find one.

  Could not find a package configuration file provided by "gazebo" with any
  of the following names:

    gazeboConfig.cmake
    gazebo-config.cmake

  Add the installation prefix of "gazebo" to CMAKE_PREFIX_PATH or set
  "gazebo_DIR" to a directory containing one of the above files.  If "gazebo"
  provides a separate development package or SDK, be sure it has been
  installed.

-- Configuring incomplete, errors occurred!
See also "/ros_ws/build_isolated/gazebo_ros/CMakeFiles/CMakeOutput.log".
See also "/ros_ws/build_isolated/gazebo_ros/CMakeFiles/CMakeError.log".
<== Failed to process package 'gazebo_ros': 
  Command '['/opt/ros/env.sh', 'cmake', '/ros_ws/src/gazebo_ros', '-DCATKIN_DEVEL_PREFIX=/ros_ws/devel_isolated/gazebo_ros', '-DCMAKE_INSTALL_PREFIX=/opt/ros', '-DCMAKE_BUILD_TYPE=Release', '-G', 'Unix Makefiles']' returned non-zero exit status 1

Reproduce this error by running:
==> cd /ros_ws/build_isolated/gazebo_ros && /opt/ros/env.sh cmake /ros_ws/src/gazebo_ros -DCATKIN_DEVEL_PREFIX=/ros_ws/devel_isolated/gazebo_ros -DCMAKE_INSTALL_PREFIX=/opt/ros -DCMAKE_BUILD_TYPE=Release -G 'Unix Makefiles'
```

gavanderhoorn, 2018-04-05:
This was also reported by @ipa-nhg-hd in robust-rosin/b826eae_cob_robots#1.

I'm a bit unsure as to why this is now a problem for two reasons:

 1. this built in the (very recent) past for me
 1. `gazebo` is a direct dependency of `gazebo_ros`. If it can't be installed, I would expect the `rosdep install ..` step (earlier in the `Dockerfile`) to already fail

For 2: could you provide the `Dockerfile` @ChrisTimperley?

gavanderhoorn, 2018-04-05:
Ah, wait. I think I now what's going on. I'll need the `Dockerfile` to confirm it though.

> And `gazebo` is distributed via the same OSRF repository that you installed all the other pkgs from.

this is definitely true, but you would need to add that repository to `apt`'s sources for all of this to work.

I believe we opted to not do this (for now).

This would seem to be a thing we need to decide: treat Gazebo as a system dependency and *not* build it from sources, or treat it as a ROS package and make it part of `deps.rosinstall`.

I think I would favour the second option, although I'd like it if we could avoid adding that repository to all Docker images.

ChrisTimperley, 2018-04-16:
@gavanderhoorn, I'm using the generic Dockerfile: https://raw.githubusercontent.com/robust-rosin/robust/master/Dockerfile

gavanderhoorn, 2018-04-16:
I know. Summarising: if we want/need Gazebo, and don't want to build it from sources, add OSRF repository. If not, we'll need to add it to `deps.rosinstall`.

ipa-jfh, 2018-04-24:
So the issue is solved now with this commit? https://github.com/robust-rosin/robust/commit/929211d904dcfff490f420859f992af8aa786b0c

gavanderhoorn, 2018-04-24:
Should be, yes.

@ChrisTimperley: I'm wondering whether we should not make adding the OSRF repos conditional on a build-arg as well (like the switch to the old apt repos).

They are not needed for (I believe) 90% of the bugs, and they add another 'moving target' to the `Dockerfile` (ie: a contiously updated pkg repository that hosts artefacts that will be included in the image, without us being in control over which versions of those artefacts are included).

ipa-hsd, 2018-04-25:
I was able to build the workspace with this Dockerfile. Couple of modifications:
After cloning the `REPO_FORK_URL`, I had to do a `reset --hard` to switch to the commit where the bug was. Should the commit's SHA1 be an arg in bugzoo yml?

ipa-jfh, 2018-05-07:
This is solved now

closed 2018-05-07 by ipa-jfh