robust-rosin/robust #4: Decide process for discriminating between expected/unrelated build failures

ChrisTimperley, 2018-04-05:


wasowski, 2018-04-05:
I am investigating this (don't seem to be able to assign the issue to myself).

wasowski, 2018-04-05:
I can confirm that I get the same build error as you do (at least this Docker thing is stable :) ). 
The bug description shows the same error message as the original issue: https://github.com/yujinrobot/kobuki_core/issues/29 . It seems that we are not building things right :( . My gut feeling is that the build fails pretty early. 

I will try to see what commit we are building, but any ideas are welcome.  It is important that we reproduce the actual problem, not just "some build problem".

wasowski, 2018-04-05:
@gavanderhoorn : shouldn't `ecl_build` appear in the deps.rosinstall file? (it does not, but `ecl_command_line` does appear)

gavanderhoorn, 2018-04-05:
I just checked what I have here locally, and that seems to work. Or at least, I get this (which is expected):

```
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- ~~  traversing 1 packages in topological order:
-- ~~  - kobuki_ftdi
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- +++ processing catkin package: 'kobuki_ftdi'
-- ==> add_subdirectory(kobuki_core/kobuki_ftdi)
-- Could NOT find PkgConfig (missing:  PKG_CONFIG_EXECUTABLE)
-- Checking for one of the modules 'libusb'
CMake Error at /usr/share/cmake-3.5/Modules/FindPkgConfig.cmake:578 (message):
  None of the required 'libusb' found
Call Stack (most recent call first):
  kobuki_core/kobuki_ftdi/CMakeLists.txt:7 (pkg_search_module)

-- Checking for one of the modules 'libftdi'
CMake Error at /usr/share/cmake-3.5/Modules/FindPkgConfig.cmake:578 (message):
  None of the required 'libftdi' found
Call Stack (most recent call first):
  kobuki_core/kobuki_ftdi/CMakeLists.txt:8 (pkg_search_module)

CMake Warning at /opt/ros/kinetic/share/catkin/cmake/catkin_package.cmake:166 (message):
  catkin_package() DEPENDS on 'libusb' but neither 'libusb_INCLUDE_DIRS' nor
  'libusb_LIBRARIES' is defined.
Call Stack (most recent call first):
  /opt/ros/kinetic/share/catkin/cmake/catkin_package.cmake:102 (_catkin_package)
  kobuki_core/kobuki_ftdi/CMakeLists.txt:10 (catkin_package)

CMake Warning at /opt/ros/kinetic/share/catkin/cmake/catkin_package.cmake:166 (message):
  catkin_package() DEPENDS on 'libftdi' but neither 'libftdi_INCLUDE_DIRS'
  nor 'libftdi_LIBRARIES' is defined.
Call Stack (most recent call first):
  /opt/ros/kinetic/share/catkin/cmake/catkin_package.cmake:102 (_catkin_package)
  kobuki_core/kobuki_ftdi/CMakeLists.txt:10 (catkin_package)
```

However, as you can see, I'm only building `kobuki_ftdi`:

```
Using CATKIN_BLACKLIST_PACKAGES: kobuki_core;kobuki_dock_drive;kobuki_driver
```

That is most likely the issue here. `rosinstall_generator` works with *packages*, not repositories. So the `.rosinstall` files will only contain the dependencies for one specific package, in this case `kobuki_ftdi`.

gavanderhoorn, 2018-04-05:
Btw: @ChrisTimperley: I did have to add `git` to the "install bootstrap dependencies" part of the `Dockerfile`, as it's needed by `wstool` later on.

Not sure why that suddenly needs to be added. Perhaps some transitive dependency got cleaned up.

wasowski, 2018-04-06:
@gavanderhoorn I am, of course, lost in all the build system magic.  So how and where do you involve this USING clause (in your quotation above), and how we can make that automatic? (as far as I understand you, the present build call in the docker file attempts to build too much).

gavanderhoorn, 2018-04-06:
That is not a clause. That is output from `catkin_make`. See the same line (but for the whitelist) in @ChrisTimperley's copy-pasted output.

I think the easiest way to go about this is to just build the `kobuki_ftdi` package, and not all the others.

That package is also the one the bug was reported for, so it would seem to make sense.

gavanderhoorn, 2018-04-06:
You should probably make [this line](https://github.com/robust-rosin/robust/blob/47c7e3add1447c48a1e0f11310a9e84a013f1fe7/kobuki/kobuki.bugzoo.yml#L13) read `kobuki_ftdi` instead of `kobuki_core`.

ChrisTimperley, 2018-04-06:
Fixed by restricting the build to `kobuki_ftdi`.

closed 2018-04-06 by ChrisTimperley