robust-rosin/robust #5: Failing to use bugzoo with eed104d

wasowski, 2018-04-05:
The first time I try to use bugzoo so please forgive fumbling:

```
ros@Gorce:~/work/robust$ bugzoo bug build robust:eed104d
building bug: robust:eed104d
Building image: rosin/robust:eed104d
Step 1/21 : ARG ROS_DISTRO
Traceback (most recent call last):
  File "/home/ros/.local/bin/bugzoo", line 11, in <module>
    sys.exit(main())
  File "/home/ros/.local/lib/python3.5/site-packages/bugzoo/cli/__init__.py", line 532, in main
    args.func(args)
  File "/home/ros/.local/lib/python3.5/site-packages/bugzoo/cli/__init__.py", line 487, in <lambda>
    cmd.set_defaults(func=lambda args: build_bug(rbox, args.bug, args.force))
  File "/home/ros/.local/lib/python3.5/site-packages/bugzoo/cli/__init__.py", line 101, in build_bug
    rbox.bugs.build(bug, force=force)
  File "/home/ros/.local/lib/python3.5/site-packages/bugzoo/mgr/bug.py", line 83, in build
    quiet=quiet)
  File "/home/ros/.local/lib/python3.5/site-packages/bugzoo/mgr/build.py", line 124, in build
    raise ImageBuildFailed(name, log)
bugzoo.core.errors.ImageBuildFailed: rosin/robust:eed104d
```

(this is after installing bugzoo with pip3 and adding a local copy of robust)

@ChrisTimperley to the rescue?

wasowski, 2018-04-05:
Hmm. It might be that my docker is too old? I tried to call the build manually (by modifying a local copy of /build.sh)
```
ros@Gorce:~/work/robust$ docker build --build-arg UBUNTU_VERSION=xenial --build-arg ROS_DISTRO=kinetic --build-arg IS_BUILD_FAILURE=yes -f kobuki/eed104 d.L3/.Dockerfile -t cool kobuki/eed104d.L3
Sending build context to Docker daemon 9.216 kB                             
Step 1/21 : ARG ROS_DISTRO                                                  
Please provide a source image with `from` prior to commit                   
```
The message is slightly more informative (and likely the same problem is causing it).  What is the version of docker that is required?

gavanderhoorn, 2018-04-05:
hm, what is the output of `docker --version`?

Being able to use `ARG` in `FROM` is relatively new, yes.