robust-rosin/robust #6: add more users

wasowski, 2018-04-05:
We need to add Claus Brabrand  (with his ITU email).  I could not figure out how to add people.

wasowski, 2018-04-05:
@gavanderhoorn tagging you -as I don;t seem to be able to assign folks, even in the issues I created.

gavanderhoorn, 2018-04-05:
To get ppl into the org, we need their github accounts.

gavanderhoorn, 2018-04-05:
?

gavanderhoorn, 2018-04-05:
I can actually invite via email.

Would you prefer I do that?

See https://help.github.com/articles/inviting-users-to-join-your-organization/

gavanderhoorn, 2018-04-05:
There is already an outstanding invitation for @brabrand to join this organisation.

Claus would just have to accept it.

wasowski, 2018-04-05:
ok. I closed (and forgot to comment).  I realized that this is open already anyways, and it is unlikely that he needs anything more than read access.  Apologies for abrupt red signs.

gavanderhoorn, 2018-04-05:
I've changed default permissions for all members of the organisation to `write`. That is one below `admin`. I believe this should be sufficient, seeing as members are already `admin` on the repositories they clone (and work on).

gavanderhoorn, 2018-04-05:
If @brabrand cannot find the invitation any more, he could open [github.com/robust-rosin](https://github.com/robust-rosin) in a browser and the invite shouls how up again.