robust-rosin/robust #9: How to Proceed With ROS Core Packages

git-afsantos, 2018-04-11:
I am currently working on a bug on `roscpp`, one of the core packages from `ros_comm`. After a few experiments building it on ym workspace, I quickly realized that building these core packages messes up with the current ROS installation. ROS gives precedence to packages on your workspace, so when you launch any node or tool that depends on `roscpp` afterwards, it will try to use the copy you are working on, instead of your system's `roscpp`.

This is obviously undesirable when you are working on your own machine/workspace, but it might not be an issue within Docker images, since their sole purpose is to reproduce the bug and its fix.

In any case, I just wondered if it was worth going the extra length of changing the package name in these cases (e.g. `roscpp_ca23e58`), just so we can guarantee a working, stable ROS installation, regardless of the PUT belonging to `ros_comm`.

gavanderhoorn, 2018-04-11:
For me personally, this was one of the reasons to start looking at Docker to package all of this: absolute isolation of the reproduced system state from the 'host'.

I wouldn't start renaming the packages: it's very easy to miss a reference and it means more work.

Note btw that you can create as many workspaces as you want / need: I have 56 at the moment. None of those share any packages. The first thing to do would be to not have a `source $CATKIN_WS/..` in your `.bashrc`, but only `source` the workspace you need.

gavanderhoorn, 2018-04-11:
@git-afsantos wrote:
> Up until now, all I ever read was *"avoid multiple workspaces, use overlays instead"*.

well, if you actually *want* to make use of packages in other workspaces, that is good advice. Disparate workspaces obviously make it impossible to share resources :)

But seeing as that is exactly what we're after here, I'd say we can just have separate workspaces and source the one you're going to work with.

Note btw that overlaying also *always* involves multiple workspaces, so I'm not sure "avoid multiple workspaces, use overlays instead" sec makes much sense?

gavanderhoorn, 2018-04-11:
@git-afsantos wrote:
> This sounds obvious, now that you wrote it down. I wish this kind of information was explicit somewhere in the ROS Wiki or ROS Answers, though. 

in all my answers on ROS Answers, I always mention that if you want / need multiple workspaces, avoid auto-sourcing any one in particular.

With all questions about "how do I avoid having to source ..", I always answer: add it to your `.bashrc`, *unless* you want/need multiple workspaces.

But I agree with you that it's left slightly implicit. I guess the idea is that this sort of thing becomes 'easy' once you understand workspaces and overlaying.

gavanderhoorn, 2018-04-11:
@git-afsantos wrote:
> I was just curious about everyone's opinion on this

Well, this was mine :)

let's see whether some of the other collaborators have other ideas.

closed 2018-04-21 by git-afsantos