robust-rosin/robust #11: Use repo tags to automate parts of the build process

ChrisTimperley, 2018-04-16:


wasowski, 2018-04-19:
doc/testcase.md has the tag names now (just added) and the entire branching procedure.  I still did not describe the scripts that you will have to switch branches, etc. (coming). I thought this might be useful for you to know, though.

ChrisTimperley, 2018-04-23:
This is pretty much implemented. We can discuss some of the design decisions in the meeting. `fix` and `unfix` are now idempotent; under the hood, each of those scripts calls `switch` to change between `robust_*` tags in the repository under test. There is also an automatically generated patch file `fix.patch`. Unlike in the previous scripts, `fix.patch` is no longer used to switch between the buggy and fixed versions -- it's just there to make it easier for the user to see the fix.