robust-rosin/robust #12: Make names of bug collections consistents with repos?

wasowski, 2018-04-18:
The tf2 bugs are now in tf2/ and the yml file is tf2.bugzoo.yml.  It is slightly confusing, as the repository name is geometry2.  Shoudl we name all collections after the original repository names? so it should be geometry2/ and geometry2.bugzoo.yml ? This will make the link between the forks in the organization and the rest of the structure more clear.

gavanderhoorn, 2018-04-18:
This makes sense to me.

I haven't checked, but what about bugs that need multiple repositories?

ChrisTimperley, 2018-04-18:
Sounds good to me.

wasowski, 2018-04-18:
I think all bugs are harvested only from one of the eight projects - one repo per project (I am ignoring 'confidential').  Of course, some will need more repos for reproduction, but these are just dependencies, aren't they.  In fact, each bug needs hundreds of these.