robust-rosin/robust #13: Unclear instructions on test case README

git-afsantos, 2018-04-18:
Specifically, this point is confusing me.

> 5. Create a file README.bug_witness at the root of this repo, containing a URL to your repo, plus any important comments about reproduction you think you need to make. This is so that we can find your repo if we wanted to fix something in this bug - this link will land in our patch file, so we can trace from the database to this repo.
>
> Of course you should commit to this repo, to the bug_witness branch as you go so that you have version control for your test development work.

Which repository is *this repo* and which is *your repo*? It is my understanding that we should develop the test on the *hash_project* repository. Is this where this `README.bug_witness` should be created, or is it within the *project/hash* directory, within this repository (*robust-rosin/robust*)?

git-afsantos, 2018-04-19:
Sounds good to me. I remember now that we already discussed that idea a few weeks ago. :+1:

closed 2018-04-19 by git-afsantos

wasowski, 2018-04-19:
done now in commit 1e01d60c47f87e4d9c92a77f14993fa6711f102f and commit 2cd7c8d30f359561cb3ea59ee51f9cedf667e730 .
I also moved the files to doc/ as the root directory started to be cluttered.

wasowski, 2018-04-19:
@git-afsantos : you may like to know that I have revised the entire procedure for the test case (the last part of doc/testcase.md now).  It is far from perfect, and the Docker part has still some mess in there, but it is way better than what was there before.

git-afsantos, 2018-04-19:
Thanks, @wasowski ! That will certainly help making sure that we all follow the same process. I believe I have a test case ready, it is just missing these setup and infrastructure details.