robust-rosin/robust #24: Bugzoo v2.1.2 bug

ipa-jfh, 2018-05-04:
Bugzoo returns an error

```bash
$ bugzoo source add robust https://github.com/robust-rosin/robust
Traceback (most recent call last):
  File "/home/jfh/bugzoo_venv/bin/bugzoo", line 7, in <module>
    from bugzoo.cli import main
  File "/home/jfh/bugzoo_venv/lib/python3.5/site-packages/bugzoo/__init__.py", line 2, in <module>
    from bugzoo.client import Client
  File "/home/jfh/bugzoo_venv/lib/python3.5/site-packages/bugzoo/client/__init__.py", line 1, in <module>
    from .api import APIClient
  File "/home/jfh/bugzoo_venv/lib/python3.5/site-packages/bugzoo/client/api.py", line 1, in <module>
    from typing import Optional, Any, NoReturn
ImportError: cannot import name 'NoReturn'
```
even without an argument

```bash
$ bugzoo
Traceback (most recent call last):
  File "/home/jfh/bugzoo_venv/bin/bugzoo", line 7, in <module>
    from bugzoo.cli import main
  File "/home/jfh/bugzoo_venv/lib/python3.5/site-packages/bugzoo/__init__.py", line 2, in <module>
    from bugzoo.client import Client
  File "/home/jfh/bugzoo_venv/lib/python3.5/site-packages/bugzoo/client/__init__.py", line 1, in <module>
    from .api import APIClient
  File "/home/jfh/bugzoo_venv/lib/python3.5/site-packages/bugzoo/client/api.py", line 1, in <module>
    from typing import Optional, Any, NoReturn
ImportError: cannot import name 'NoReturn'
```

Installed version

```bash
$ pip show bugzoo
Name: bugzoo
Version: 2.1.2
```

Temporary solved by

```bash
pip3 install bugzoo==2.0.15
```

ChrisTimperley, 2018-05-04:
Hi Jonathan,

Can you try to install `mypy` and `mypy-extensions`? I realise that `NoReturn` is defined in `mypy-extensions` but that it's not included in the requirements listed in `setup.py`. Once we're happy that's the bug, I'll fix `setup.py` and release 2.1.3.

ipa-jfh, 2018-05-07:
Hi Chris,
I tested to install this two libs, but I still get the same error.

ChrisTimperley, 2018-05-07:
Do you mind trying this inside the virtual environment and dumping the output here?
```
$ pip install --upgrade mypy-extensions>=0.3.0
$ pip list
```

ChrisTimperley, 2018-05-08:
Thanks, @ipa-jfh. It looks like the issue is that `NoReturn` was added in Python 3.5.4. I'll figure out a way to recover compatibility with early versions of Python 3.5 without causing mypy to scream.

ChrisTimperley, 2018-05-15:
A fix to this bug was added in BugZoo 2.1.6.

closed 2018-05-15 by ChrisTimperley