robust-rosin/robust #26: Version of dependency not correct since its release is outdated

ipa-jfh, 2018-05-07:
https://github.com/robust-rosin/robust/commit/61bc23ba260313dd74733f145207c2267939b130 -> changed version from `0.6.8` to `0.6.9`

We had to change the version of one dependency since at occurrence of the bug the released version of the dependency was not up to date to the latest commit.

This are the dates of the bug:
`on Nov 10, 2016: robust_fixed_released    b826eae`
`on Nov 2, 2016: robust_buggy_released    6ca6d50`

The deps.rosinstall referenced to the
`on Oct 10, 2016: 0.6.8  fe15fd3 `

but the bug is depending to the following commit 

`on Nov 2, 2016 https://github.com/ipa320/schunk_modular_robotics/commit/779ce503e0f6b868f04253a75590f7e10c086427#diff-4c3ad0877d980bf914fa8fb2d7cb30b8`

which has been released roughly a year later
`on Jul 17, 2017: 0.6.9  0a78223`

It is an easy fix for us but just letting you know @gavanderhoorn

gavanderhoorn, 2018-05-07:
Was the bug reporter using a from-source install of `schunk_modular_robotics`? Or was the commit used to generate the `.rosinstall` file not the correct one?

gavanderhoorn, 2018-05-07:
I can't find the build log any longer on Travis for a build around that time, but I'm not sure I understand how rosdistro@96a8e741 does not point to the correct version of `schunk_modular_robots` unless CoB was using a from-source install for that pkg.

Or am I misunderstanding something here?

gavanderhoorn, 2018-05-07:
If it was a from-source install, should we update `deps.rosinstall` to point to the source repository and spec a commit instead of a version?

ipa-jfh, 2018-05-16:
They first "it" is referring to the magical solution to get the latest commit (instead of version) of all the dependencies. But as I understood now this is not possible and also not reasonable. My last comment can thus be ignored.

a)
> Or we could retain it in deps.rosinstall and just make the entry point to the source repository.

> ... point to the source repository and spec a commit instead of a version?

So what I did (https://github.com/robust-rosin/robust/commit/61bc23ba260313dd74733f145207c2267939b130) is kind of similar to this solution but I should have used the newer commit and not simply the newer version (`0.6.9`). This would make it consistent with future similar cases. This approach would be very easy to realize.

b)
> only in case of developers using from-source builds remove a released dependency from deps.rosinstall and add (somehow) add it as a source dependency

I would for now suggest to use the a) solution, because with b) we might need to introduce sth like another `.rosinstall` file.

gavanderhoorn, 2018-05-16:
@ipa-jfh wrote:
> So what I did (https://github.com/robust-rosin/robust/commit/61bc23ba260313dd74733f145207c2267939b130) is kind of similar to this solution but I should have used the newer commit and not simply the newer version (`0.6.9`).

yes, I would say using the hash instead of the version would be best.

> I would for now suggest to use the a) solution, because with b) we might need to introduce sth like another `.rosinstall` file.

Now that we don't have files for the PUTs that is true.