robust-rosin/robust #30: Migrate to single-fork-many-branch

gavanderhoorn, 2018-05-16:
We'll need to

 - [ ] rename the current forks (as they now have a bug id as part of their name)
 - [ ] update the various bugzoo yaml files to point to branches/tags

@ChrisTimperley: there's also things for the `Dockerfile` of course, so please feel free to add those to the list (just edit my post).

gavanderhoorn, 2018-05-16:
I'll also delete [b166c93_kobuki_core](https://github.com/robust-rosin/b166c93_kobuki_core) and need to update #19 to use `eed104d_kobuki_core` instead.

wasowski, 2018-05-18:
I am still confused so perhaps a silly question. Why delete `b166c93*` repo and use `eed104*` repo? Shouldn't the `eed104d_kobuki_core` be also renamed to `kobuki core` and the bugzoo files should use that name?

gavanderhoorn, 2018-05-18:
> Why delete `b166c93*` repo

because it is a duplicate of `eed104d_kobuki_core`, which I made using the process described in #18, and since we decided we would not mirror but instead would use single clones with multiple branches. 

`b166c93*` is thus not needed any longer.

> Shouldn't the `eed104d_kobuki_core` be also renamed to `kobuki core` and the bugzoo files should use that name?

Yes, that is what "rename the current forks" is intended to capture.

wasowski, 2018-05-18:
ah ok.

ChrisTimperley, 2018-05-26:
@wasowski @gavanderhoorn is there any chance that you could rename the forked repos, or alternatively, could you give me permission to make changes to the forked repos?

gavanderhoorn, 2018-05-27:
@ChrisTimperley wrote:
> you could rename the forked repos

done.

commit 7c778dac (2018-05-27): Updated Dockerfile and BugZoo files to use one-fork-many-branch model (fixes #30) (#48)

Updated Dockerfile and BugZoo files to use one-fork-many-branch model (fixes #30)
closed 2018-05-27 by ChrisTimperley