robust-rosin/robust #32: Removing temporaries? (migrated from gavanderhoorn/rosin_bug_hunt_l3)

gavanderhoorn, 2018-05-18:
https://github.com/gavanderhoorn/rosin_bug_hunt_l3/issues/5

Just linking this here, so the discussion doesn't get lost.

Summary: the `Dockerfile` ([this version](https://github.com/robust-rosin/robust/blob/689b9445a61bd1d8cdc10ff2bf4fde0221dbec3d/Dockerfile)) currently deletes `build` and `devel` folders that may be needed by some tests / in some cases.

We need a decision on this.

gavanderhoorn, 2018-11-24:
We're still doing this ([here](https://github.com/robust-rosin/robust/blob/16ec82e60bf2fb4e204a20c13c3546056cfa0d64/Dockerfile#L143-L146)).

ChrisTimperley, 2018-11-25:
I have no objection to retaining those directories, although it may end up creating very bloated images. One possibility is that we add another build-time flag to the Dockerfile that can be used to toggle whether or not those directories should be removed.

gavanderhoorn, 2018-11-26:
I'm not sure retaining those directories is necessary.

They're only used when running tests, but I don't know whether users of the images would like to do that. It would also only affect the packages of the underlay that get installed into `/opt/ros`.

Tests are typically only run for packages in your workspace -- which would be the overlay workspace we create after installing the underlay stuff.