robust-rosin/robust #34: Build image for  6c13c78

wasowski, 2018-05-21:
@gavanderhoorn I have this bug in branch  `6c13c78` of the robust repo.  There is a rudimentary bug file in `geometry2/6c13c78.bug`. It is not finished, but I feel very uneasy to commit to the description of this bug before I actually managed to reproduce the crash.  There is so little description in the PR, and so many unrelated changes in the commit, that there is a huge risk that I misdiagnosing.  

Can you try to see whether we can build the rosinstall file for this (and then @ChrisTimperley could try to build the container?).  I am almost clear on how to write the test case, so I would be eager to try.

wasowski, 2018-05-21:
BTW. When you have the file, commit it to the branch, and I will merge, when I have a test case (and know for sure what the bug is)

gavanderhoorn, 2018-05-21:
I just checked, and what I 'feared' was confirmed by this traceback:

```
AssertionError: Unable to handle 'index' format version '1' anymore, please update your 'index' file to version '2' or '3'
```

Everything older than 24th of January 2014 will need some additional work. Index files are too old.

gavanderhoorn, 2018-05-21:
The rosdistro commit, in any case: 46599cc4ea32c0d2ac589106ed2a3c51fa5a2fc4 (ros/rosdistro@46599cc4ea32c0d2ac589106ed2a3c51fa5a2fc4).

wasowski, 2018-05-21:
Ok let's leave it open.  I thought the cut off date was a bit earlier.  I will work forward. Probably just a handful more before I cross to January 2014.

gavanderhoorn, 2018-05-21:
@wasowski wrote:
>  I thought the cut off date was a bit earlier.

well, it is 4 years in the past, which is already almost a 'century' :)

But I'll see if I can do something about that.

wasowski, 2018-05-22:
I'd say it is not the first priority.  The next one (that I am already working on) is already fixed after the cut off point.  So I will ask you to see what surprises this one might hide (soon).

gavanderhoorn, 2018-11-12:
This is now possible with https://github.com/rosin-project/rosinstall_generator_time_machine/pull/8.

Command line:

```
./rosinstall_generator_tm.sh "2013-08-28T23:03:29Z" hydro --deps --deps-only tf2
```

timestamp retrieved using:

```
./get_issue_creation_date.py https://github.com/ros/geometry2/pull/25
```

gavanderhoorn, 2018-11-25:
@wasowski: did you remove the branch you're referring to in https://github.com/robust-rosin/robust/issues/34#issuecomment-390672568?

gavanderhoorn, 2018-11-25:
nvm. Just found it.

gavanderhoorn, 2019-02-20:
See #216.