robust-rosin/robust #36: Create image for geometry2/12605ab

wasowski, 2018-05-22:
I dared to commit this one to master, as I am reasonably certain of the description (even if not finished).  This one is from Spring 2015, so we have some hope for the time machine working, despite the current limitations of our time travel technology :)

I think that I can reproduce it (well, I might need to produce a SIGINT against some core node in the test, but at least I know the direction what to think about unlike for many others :) )

gavanderhoorn, 2018-05-22:
So the fix is from `2015-03-12T01:33:01Z`, but the actual report is from `2013-09-17T13:34:49Z`. Are we ok with a `.rosinstall` file for 2015?

gavanderhoorn, 2018-05-22:
Btw: could I ask you to link to the bug description file when asking for a `.rosinstall` file? :) makes it easier for me to find the issue/pr link.

wasowski, 2018-05-22:
Sure. I will try.  The mode of work for tf2 is a bit different, so I am still learning.  I think it makes sense to try to test them sooner than later, thus I make these issues.  This is because, I still remember what they are about (unlike the UR bugs that are completely alien to me now, so they can just as well wait, or be taken by someone else...). 

Re the date: yes this is my assumption that we should be able to reproduce it based on the time of the parent of the fixing commit, not on the reporting the issue time.  I kind of expect that this was still a problem, when they were fixing this.

gavanderhoorn, 2018-05-22:
See e15feaae6d2f388426d6ae71cd06c56b4f1829c7.

wasowski, 2018-05-22:
btw. this was super fast.  It took only 5 minutes?

gavanderhoorn, 2018-05-22:
It looked like 5 mins for you, I have a time-machine :sunglasses:

commit b9ba3e82 (2018-05-25): added bugzoo data for 12605ab (fixes #36) (#41)
closed 2018-05-25 by ChrisTimperley