robust-rosin/robust #40: Document required Docker version explicitly

gavanderhoorn, 2018-05-25:
I keep running into the `invalid reference format` error message on various hosts.

We should document the version of `docker` that is required in the readme.

Perhaps even provide a `check_setup.sh` script or something that verifies whether things have been installed / setup / configured correctly.

git-afsantos, 2018-05-25:
Probably not the answer, but I just ran into the same error message with the command you provided in #20 .

In this case, however, the problem was in the `build-arg`s given. You listed `ROS_DISTRO` twice, and one of them, with the value `xenial` should have been `UBUNTU_VERSION` instead.

gavanderhoorn, 2018-05-25:
Yes, you're right. I'd actually edited the `Dockerfile` to not require an `ARG` for the `FROM` line, so I didn't notice.

But we still should document the absolute minimum version of `docker` ;)

git-afsantos, 2018-05-25:
True. After having read an older issue by @wasowski , I went ahead and directly installed Docker 18 too, but an earlier version might be enough.