robust-rosin/robust #42: bugzood files 'everywhere'?

gavanderhoorn, 2018-05-25:
This should really be opened on the `bugzoo` issue tracker, but as we have @ChrisTimperley here, thought I'd do it like this.

Every time I run `bugzoo`, I end up with a `bugzood` file in the `cwd`.

This started happening after I did a `pip install -U bugzoo` today.

Has some debugging code been left enabled in the latest release @ChrisTimperley?

ChrisTimperley, 2018-05-25:
In the latest release, I tweaked BugZoo to always write its logs to `bugzood`, whereas previously it only used to dump its logs in the event of a crash. It can be quite useful for debugging, but if you think it's an annoyance, then I'm happy to revert it to the previous behaviour.

ChrisTimperley, 2018-05-25:
Alternatively, BugZoo could keep logging each session to `~/.bugzoo/logs`.

gavanderhoorn, 2018-05-25:
> Alternatively, BugZoo could keep logging each session to `~/.bugzoo/logs`.

I would be in favour of that.

gavanderhoorn, 2018-05-25:
Let's close this one here, as you're already tracking this over at squaresLab/BugZoo#219.

closed 2018-05-25 by gavanderhoorn