robust-rosin/robust #44: Single bugzoo yaml per 'program', or multiple?

gavanderhoorn, 2018-05-25:
Right now we have a single `.bugzoo.yml` per repository / program.

I'm wondering if it would be ok to have one per bug.

If that is ok (for BugZoo as well as us humans), then I think I can automate things a little more.

Opinions?

ChrisTimperley, 2018-05-25:
BugZoo is fine with any approach: it just scans for all `.bugzoo.yml` files in the repository.

I'm in favour of one per bug.

gavanderhoorn, 2018-05-27:
I've started with this in #49 and #50.

gavanderhoorn, 2018-05-27:
@ChrisTimperley: what is the exact pattern that BugZoo scans for?

Would having just a `bugzoo.yml` also work?

ChrisTimperley, 2018-05-27:
@gavanderhoorn it scans for all files ending in `.bugzoo.yml`.

gavanderhoorn, 2018-05-30:
I've just ended up with a single `manifest.bugzoo.yml` in the bug dir.

closed 2018-05-30 by gavanderhoorn