robust-rosin/robust #45: Motoman bugs: MotoROS is untestable (at least as L3)

gavanderhoorn, 2018-05-25:
Example: ``: ros-industrial/motoman#138.

These are bugs in the MotoROS part of the driver, which only runs on the industrial robot controller, not on the ROS side. The code targets VxWorks / MotoPlus SDK, so cannot be compiled on regular Linux/Ubuntu (well, it can, but it's not intended to).

Does it make sense to create images for these kind of bugs?

@wasowski, @brabrand?

gavanderhoorn, 2018-06-09:
The following bugs are also in MotoROS and thus unreproducable using our standard approach:

 - https://github.com/robust-rosin/robust/blob/master/motoman/292b5cc.bug
 - https://github.com/robust-rosin/robust/blob/master/motoman/377d7be.bug
 - https://github.com/robust-rosin/robust/blob/master/motoman/3f260cb.bug
 - https://github.com/robust-rosin/robust/blob/master/motoman/4089660.bug
 - https://github.com/robust-rosin/robust/blob/master/motoman/acd5eab.bug

gavanderhoorn, 2018-07-03:
Same for UR actually: #53.

wasowski, 2018-07-03:
Yes, let's mark then as unreproducable.  I suggest to do L2 for them, if we are asked by reviewers.