robust-rosin/robust #46: Empty 'ROS_DISTRO'?

gavanderhoorn, 2018-05-25:
Not sure what is going on, just logging this for now.

While building an image for `f8d175b` (universal_robot), I see the following in the build log:

```
-- Installing: /opt/ros/lib/pkgconfig/tf2_ros.pc
-- Installing: /opt/ros/share/tf2_ros/cmake/tf2_rosConfig.cmake
-- Installing: /opt/ros/share/tf2_ros/cmake/tf2_rosConfig-version.cmake
-- Installing: /opt/ros/share/tf2_ros/package.xml
-- Installing: /opt/ros/lib/libtf2_ros.so
```
and so on.

Notice the `/opt/ros` path. There should be a ROS version suffix there (such as `indigo` or `kinetic`), but it appears to not be present.

The BugZoo metadata that configured this build:

```yaml
version: '1.0'

blueprints:
  - type: docker
    tag: robustrosin/robust:f8d175b
    file: ../Dockerfile
    context: f8d175b
    arguments:
      IS_BUILD_FAILURE: "yes"
      USE_APT_OLD_RELEASES: "no"
      UBUNTU_VERSION: "trusty"
      ROS_DISTRO: "indigo"
      CATKIN_PKG: "ur_kinematics"
      REPO_FORK_URL: "https://github.com/robust-rosin/universal_robot"

bugs:
  - name: robust:f8d175b
    image: robustrosin/robust:f8d175b
    program: universal_robot
    dataset: robust
    languages:
      - cpp
    source-location: /ros_ws/src
    test-harness:
      type: empty
    compiler:
      type: catkin
      workspace: /ros_ws/src
      time-limit: 300
```

That looks alright to me (I'm even using the quotes introduced in b94590e8 by @ChrisTimperley.

Somehow [this line](https://github.com/robust-rosin/robust/blob/b94590e89767b1337cd74f18fc4e9df8fd3884c7/Dockerfile#L114) does not set the install target correctly (or at least: it gets set to just `/opt/ros`).

gavanderhoorn, 2018-05-27:
Running into this again while building #49.

ChrisTimperley, 2018-05-27:
I think that I may have just fixed this in efbbf147e69bb5eb7fbaee7034d5b2c0df735308. Let me know if the problem persists.

gavanderhoorn, 2018-05-30:
Looks like this is ok now.

I'll check later with some new builds.

closed 2018-05-30 by gavanderhoorn