robust-rosin/robust #53: UR bugs: bugs in urscript are unreproducable

gavanderhoorn, 2018-06-09:
The following bugs are issues in urscript programs:

 - https://github.com/robust-rosin/robust/blob/master/universal_robot/1e0904c.bug
 - https://github.com/robust-rosin/robust/blob/master/universal_robot/359a2e9.bug
 - https://github.com/robust-rosin/robust/blob/master/universal_robot/cda133d.bug

As we don't have any interpreter for URScript, we cannot really 'reproduce' them.

gavanderhoorn, 2018-07-03:
Ping?

this is not dependend on the time machine, but in general this is unreproducable, at least L3.

gavanderhoorn, 2018-07-03:
related: #45.

wasowski, 2018-07-03:
Yeah, I tend to agree.  At L3 it is impossible, so we should probably somehow list them as "done" (blacklisted or whatever). If we fancy this, we may do L2 for them, or just not do them.