robust-rosin/robust #54: Kobuki bugs: 1c141a5.bug is too old for current time machine

git-afsantos, 2018-06-13:
I just started working on this bug, and after a while realised I could not build the `rosinstall` file. As clarified by @gavanderhoorn this bug is too old for the current time machine date threshold.

Original issue: https://github.com/rosin-project/rosinstall_generator_time_machine/issues/2

git-afsantos, 2018-06-13:
Just for the record, `3c7f92a.bug` is also too old.

git-afsantos, 2018-06-14:
By the way, this just occurred to me, should we create a directory within each project directory to group the bugs that are too old?
This way we do not stumble upon the same bugs over and over while selecting a new bug to work on.

On the other hand, it might not be worth it for all projects.
The same logic could be applied to unreproducible bugs.

gavanderhoorn, 2018-06-14:
Sounds like a good idea to me.

gavanderhoorn, 2018-06-14:
Wow.

Just noticed your updated OP.

Isn't that just about every kobuki bug?

I'm a bit surprised, as I know for a fact that at least *some* of the kobuki bugs should work.

Could you check that your clone of `rosdistro` is not in an 'unclean' state? The script isn't very robust against failures/weird things right now, so I wouldn't be surprised if it fails for bugs it should work for.

git-afsantos, 2018-06-14:
> Isn't that just about every kobuki bug?

It is most of them, yes.
I added bugs which seemed reproducible and for which the buggy commit is pre-2014. I have not tried all these yet, but I will give a shot at some of them and will let you know.

git-afsantos, 2018-06-16:
I have now tested all bugs in the list above, and annotated the respective errors.

wasowski, 2018-07-05:
I am also creating separate issues for Gijs about these.  Should I stop doing that?

gavanderhoorn, 2018-07-05:
I think this list is ok for now.

If there are any special properties of bugs that need to be clarified a separate issue could be appropriate.