robust-rosin/robust #55: Empty rosinstall file for kobuki:5ee65b0

git-afsantos, 2018-06-16:
This is for bug `kobuki:5ee65b0`, with creation date of Aug. 2014 (should be using Indigo, judging by the date). The process seems to go well, using the following command.

```bash
$ time_machine https://github.com/yujinrobot/kobuki_desktop/issues/32 5ee65b0 indigo kobuki_dashboard deps_5ee65b0.rosinstall
```

The bug occurs in the `kobuki_dashboard` package, and I believe this last output message hints at the problem.

> The following not released packages/stacks will be ignored: kobuki_dashboard
> No packages/stacks left after ignoring not released

gavanderhoorn, 2018-06-16:
Hm, interesting. Could you copy paste the complete output of the script (without the cache building output, that is)?

gavanderhoorn, 2018-06-16:
Tracking in https://github.com/rosin-project/rosinstall_generator_time_machine/issues/5.

gavanderhoorn, 2018-06-16:
I know it gets a little confusing, but could I ask you to report issues with the time machine over at [rosinstall_generator_time_machine/issues](https://github.com/rosin-project/rosinstall_generator_time_machine/issues)? Anything that is a methodology or reproduction issue is of course fine on this tracker.

git-afsantos, 2018-06-16:
> could I ask you to report issues with the time machine over at rosinstall_generator_time_machine/issues?

Sure. I was not too sure myself whether to report it here or over there.

git-afsantos, 2018-06-17:
Closing this issue. The underlying problem is being tracked as per comment above, and for `kobuki:5ee65b0` it has been solved.

For this bug, the problem was that the GitHub issue was opened *before* the PUT was added to `rosdistro`. This makes the time machine unable to produce all dependencies using the issue report date.

The solution cannot be fully automated at the moment, and it consisted on two steps.

1. Deleting the cache and the `bughunt_5ee65b0` branch from `rosdistro` repository used by the time machine.

2. Using an explicit date, instead of an issue URL as a script argument. This has to be a manually selected date, ideally right after the date when the PUT was added to `rosdistro`.

The solution incurs a small risk of not using exactly the same dependency versions that were used when the bug was reported -- we are selecting a date slightly further into the future. In this particular case, the risk is very small, since it is only a 3 day difference.

closed 2018-06-17 by git-afsantos