robust-rosin/robust #56: deps.rosinstall for multi-package bugs

git-afsantos, 2018-06-16:
How do we handle this?

For instance, `kobuki:dd40270` affects the `kobuki/kobuki_testsuite` and `kobuki/kobuki_node` packages, but the time machine script accepts only a single PUT, as far as I understand.

For this particular case, there are multiple fixing commits, and they seem to be disjoint for each package, which should make it possible to separate this into multiple bug reports, as a last resort.

git-afsantos, 2018-06-16:
`kobuki:f95c384` is another example: a circular dependency between `kobuki/kobuki_node` and `kobuki/kobuki_capabilities`.

The fix is only applied on `kobuki_node` though (removing the dependency on `kobuki_capabilities`), so in this case we can consider `kobuki_node` as the PUT and the test is simply to detect whether it has a dependency on `kobuki_capabilities`.

gavanderhoorn, 2018-06-16:
`rosinstall_generator` (which we use eventually) does in fact support multiple pkgs as input.

I'll update the script to allow that as well. No need to split things.

gavanderhoorn, 2018-06-16:
Tracking in https://github.com/rosin-project/rosinstall_generator_time_machine/issues/4.

gavanderhoorn, 2018-11-08:
If you feel adventurous (or want to help me by testing the new version), you could check out rosin-project/rosinstall_generator_time_machine#8.

provided you know the timestamp at which you'd want your `.rosinstall` file generated, the command line would be something like:

```
$ rosinstall_generator_tm.sh \
  "ISO8601_STAMP" \
  ROS_DISTRO \
  --deps \
  --deps-only \
  kobuki_testsuite \
  kobuki_node > deps.rosinstall
```

That should result (if everything works) in a `.rosinstall` file that contains the (transitive) dependencies of both `kobuki_testsuite` as well as `kobuki_node` in `ROS_DISTRO` at the time of `ISO8601_STAMP`.

git-afsantos, 2018-11-13:
@gavanderhoorn  I am going to test this out now. What is the best approach to manually selecting a timestamp? Issue report time, fix commit time, or depends?

gavanderhoorn, 2018-11-13:
I've used issue report time, as that is typically when the issue existed.

But if there is no issue then it could be the PR itself.

git-afsantos, 2018-11-13:
@gavanderhoorn How long is building the cache supposed to take? I seem to remember you saying that it is supposed to take long (hence using caches), but now I am not sure whether it is still going or just stuck.

git-afsantos, 2018-11-13:
Turns out something *was* hanging. The process was waiting at the "enter github username" prompt, which I never got to see because of output redirection -- until I figured I should check the output file.

gavanderhoorn, 2018-11-13:
Hm. Strange.

The git pw prompt should appear on `stderr`, not `stdout`. Redirecting should not affect that.

I'll see if I can check that tomorrow.

gavanderhoorn, 2018-11-24:
@git-afsantos: with the update to the time machine this should now work. I believe @ChrisTimperley is also using this in multiple `mavros` bugs.

Can we close this?