robust-rosin/robust #59: Trouble building 0af76d5

wasowski, 2018-07-03:
I'm trying to build 0af76d5 on some tree from 2014. Hitting this:

```sh
ros@berlin:~/work/robust/geometry2$ bugzoo bug build --force robust:0af76d5
building bug: robust:0af76d5
Traceback (most recent call last):
  File "/home/ros/.local/bin/bugzoo", line 11, in <module>
    sys.exit(main())
  File "/home/ros/.local/lib/python3.5/site-packages/bugzoo/cli/__init__.py", line 532, in main
    args.func(args)
  File "/home/ros/.local/lib/python3.5/site-packages/bugzoo/cli/__init__.py", line 487, in <lambda>
    cmd.set_defaults(func=lambda args: build_bug(rbox, args.bug, args.force))
  File "/home/ros/.local/lib/python3.5/site-packages/bugzoo/cli/__init__.py", line 101, in build_bug
    rbox.bugs.build(bug, force=force)
  File "/home/ros/.local/lib/python3.5/site-packages/bugzoo/mgr/bug.py", line 83, in build
    quiet=quiet)
  File "/home/ros/.local/lib/python3.5/site-packages/bugzoo/mgr/build.py", line 89, in build
    instructions = self[name]
  File "/home/ros/.local/lib/python3.5/site-packages/bugzoo/mgr/build.py", line 28, in __getitem__
    return self.__blueprints[name]
KeyError: 'robustrosin/robust:0af76d5'
```
Any idea what could be the culprit, @ChrisTimperley ? I have put my setup on the branch 0af76d5 in the bug repo, if you would like to run this yourself to check.

PS1. For reference this is the original issue, although I think you should not need it: https://github.com/ros/geometry2/issues/35 . The bug has not been fixed yet so I think I voluntarily work on a newer tree from 2014, while @gavanderhoorn attempts to extend the range of the time machine.  I believe that the problem building is because I mis-configured something for bugzoo. 

PS2. Have not seen you on the irc #robustrosin channel today ...

gavanderhoorn, 2018-07-03:
I don't see your files on the `geometry2` fork @wasowski?

gavanderhoorn, 2018-07-03:
Ah, you meant the `robust` repository (ie: this one).

wasowski, 2018-07-03:
Fixed it.  I missed to fill in one section in the bugzoo yaml file

closed 2018-07-03 by wasowski