robust-rosin/robust #61: Build appropriate image for 0af76d5

wasowski, 2018-07-03:
The branch 0af76d5 contains the work-in-progress for 0af76d5.  This is regarding https://github.com/ros/geometry2/issues/35 . I identified the buggy commit based on the reporting time. It does not really matter, but for consistency, it would be good to rebuild the image for this commit with Hydro (starting from rosinstall files, and then the docker container). This can only be done once the time machine's range has been extended into 2013.

wasowski, 2018-07-05:
The error message I am getting (the last part of the build):
```make
Linking CXX shared library /ros_ws/devel/lib/python2.7/dist-packages/tf2_py/_tf2.so
[ 95%]
Built target tf2_py
/ros_ws/src/repo-under-test/tf2_ros/src/transform_listener.cpp: In member function ‘void tf2_ros::TransformListener::subscription_callback_impl(const TFMessageConstPtr&, bool)’:
/ros_ws/src/repo-under-test/tf2_ros/src/transform_listener.cpp:108:65: error: ‘const TFMessage’ has no member named ‘__connection_header’
     std::map<std::string, std::string>* msg_header_map = msg_in.__connection_header.get();
                                                                 ^

make[2]: *** [repo-under-test/tf2_ros/CMakeFiles/tf2_ros.dir/src/transform_listener.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....

make[1]: *** [repo-under-test/tf2_ros/CMakeFiles/tf2_ros.dir/all] Error 2

make: *** [all] Error 2

Invoking "make -j8 -l8" failed
```

gavanderhoorn, 2018-07-05:
I'll try to see what is going on.

I expect that `tf2_ros` depends on a change in `roscpp` that was not yet released into `rosdistro`, and we're building an image that contains the 'in between' state.

gavanderhoorn, 2018-07-05:
The `__connection_header` field was part of messages but removed (ros/gencpp#3) for various reasons. It was then removed in ros/geometry#52 from TF and in ros/geometry2@932c6caf for TF2.

wasowski, 2018-07-05:
I am confused how this happened but these commits are now on master, so I will delete the branch.

wasowski, 2018-07-05:
So I arrived at the pull request mentioning this commit ( ros/geometry2@932c6ca ) but this does not appear to be merged.  If it was we should take any commit after Jan 29, 2014 (https://github.com/ros/geometry2/pull/51) and try to reproduce it there.  This build error should go.  Has it actually been merged?

And second question: why this even appears? Should not we be getting roscpp that is sufficiently old? (due to rosinstall generator)

gavanderhoorn, 2018-07-05:
My last comment was not meant as a conclusion, I was simply distracted.

What I think happened is that we generated a `.rosinstall` file exactly in between two releases where one pkg (`tf2_ros`) was not yet updated, but the other (not `roscpp` btw) was not. Perhaps a day later both pkgs would be in a consistent state (ie: relationship) and everything builds.

I'll investigate some more.

wasowski, 2018-07-05:
I know the answer to the first of my questions: this was merged into indigo (I missed the note in https://github.com/ros/geometry2/pull/51). I will try to find this commit and create a new bug for it.  

The other matter still stands.  If this has never merged into hydro, I tend to believe that they were able to compile hydro somehow.  So something is fishy, or I am being naive.

gavanderhoorn, 2018-07-05:
I'm not sure I follow any more.

The PR was merged, and any version after and including `0.5.0` of `tf2_ros` should contain the fix. I don't know which commit you're using for the PUT?

wasowski, 2018-07-05:
Wait a second which PR was merged? This one (`https://github.com/ros/geometry2/pull/51`) has "closed" not merged, and there is no reference to any specific commit in the closing message by tfoote.  The message is that it was merged into indigo-devel, but when I click on the commit I am getting a review interface (not the usual UI for a merged commit).  This all confuses me.

gavanderhoorn, 2018-07-05:
If you open the `Files changed` tab, you can see that https://github.com/ros/geometry2/pull/51/commits/932c6cafd2b7959133fb32303e8c27cca3ff9e33 is part of a nr of tags and also in `melodic-devel`:

[image]

wasowski, 2018-07-05:
I am not entirely sure I understand the issue. This is what github shows to me in the PR: 
[image]
'wants to merge' and names a local branch.  I nowhere see the info that it was actually merged.  Something strange happened.  But I get your point that it was somehow merged, just not using this PR?

In any case this was in January 2014, and I think I used a much later date for this rosinstall file (because January 2014 was not supported).  So this is fishy.  I think it would make sense, when you have the new TM to rebuild it for the actual time of the PR. Perhaps there is some problem caused by me trying to reconfigure things in a custom manner.

gavanderhoorn, 2018-07-05:
@wasowski wrote:
> But I get your point that it was somehow merged, just not using this PR?

probably. You can see from the screenshot I posted (which is from the PR that you linked), that the commit was merged and that closed the PR.

Let's put this one on the backburner untill we have a TM that can generate the actual `.rosinstall` files.

wasowski, 2018-07-05:
definitely #77 should be closed first (as this is a cleaner case where this issue appears first class)

gavanderhoorn, 2019-03-06:
This was an entertaining thread to read, but I'm still not sure I understand what is going on @wasowski.

As there is no fix merged (in a PR or otherwise), I cannot setup a `bugzoo` section in the `.bug`.

There are several commit hashes mentioned in the comments here, but none actually relate to the issue in the bug report.

I've pushed an updated `.rosinstall` file and `.bug` to [geometry2/0af76d5](https://github.com/robust-rosin/robust/tree/geometry2/0af76d5). With the updated time machine we can actually generate Hydro files now, so that's where most of the changes to the `.rosinstall` come from.

gavanderhoorn, 2019-03-06:
See #255.

closed 2019-10-19 by wasowski