robust-rosin/robust #63: Error building Docker container for 5ee65b0

git-afsantos, 2018-07-04:
I tried to build this image with the following command, within `robust/kobuki/5ee65b0`.

```bash
docker build \
-t robustrosin/test-5ee65b0 \
-f ../../Dockerfile \
--build-arg UBUNTU_VERSION=trusty \
--build-arg IS_BUILD_FAILURE=no \
--build-arg USE_APT_OLD_RELEASES=no \
--build-arg ROS_DISTRO=indigo \
--build-arg CATKIN_PKG=kobuki_dashboard \
--build-arg REPO_FORK_URL=https://github.com/robust-rosin/kobuki_desktop \
--build-arg REPO_BUG_COMMIT=f04c615cd9216e1b7b6949db2745dc8fa25acc0b \
--build-arg REPO_FIX_COMMIT=5ee65b061a4a919b2a32e80087f4cb3e51e924ed \
.
```

This command failed on step 27 with the following errors.

```
Step 27/39 : RUN wstool init -j8 ${ROS_WSPACE}/src ${ROS_WSPACE}/deps.rosinstall
[...]
ERROR in config: Error processing 'tf2' : [tf2] Checkout of https://github.com/ros-gbp/geometry_experimental-release/archive/release/indigo/tf2/0.5.5-0.tar.gz version geometry_experimental-release-release-indigo-tf2-0.5.5-0 into /ros_ws/src/tf2 failed.
Error processing 'tf2_msgs' : [tf2_msgs] Checkout of https://github.com/ros-gbp/geometry_experimental-release/archive/release/indigo/tf2_msgs/0.5.5-0.tar.gz version geometry_experimental-release-release-indigo-tf2_msgs-0.5.5-0 into /ros_ws/src/tf2_msgs failed.
Error processing 'tf2_py' : [tf2_py] Checkout of https://github.com/ros-gbp/geometry_experimental-release/archive/release/indigo/tf2_py/0.5.5-0.tar.gz version geometry_experimental-release-release-indigo-tf2_py-0.5.5-0 into /ros_ws/src/tf2_py failed.
Error processing 'tf2_ros' : [tf2_ros] Checkout of https://github.com/ros-gbp/geometry_experimental-release/archive/release/indigo/tf2_ros/0.5.5-0.tar.gz version geometry_experimental-release-release-indigo-tf2_ros-0.5.5-0 into /ros_ws/src/tf2_ros failed.

The command '/bin/sh -c wstool init -j8 ${ROS_WSPACE}/src ${ROS_WSPACE}/deps.rosinstall' returned a non-zero code: 1
```

Just for reference, this is a bug that was originally giving me trouble to generate the `.rosinstall` as per #55 . I believe I ended up generating the file correctly, but considering that the error occurs on a step that uses `deps.rosinstall`, I am not so sure anymore.

gavanderhoorn, 2018-07-05:
Github seems to be having / had some prolems yesterday (and still today):

[image]

this could be a transient issue, I can download the file mentioned in the error message now.

gavanderhoorn, 2018-07-05:
Hm. Just tried the command you list and I get the same error. Downloading the file manually works fine though. Not sure what is going on.

gavanderhoorn, 2018-07-05:
The complete error seems to be:

```
ERROR [vcstools] Tarball download unpack failed: u"geometry_experimental-release-release-indigo-tf2_ros-0.5.5-0 is not a subdirectory with contents in members [u'geometry2-release-release-indigo-tf2_ros-0.5.5-0']"[/vcstools]
ERROR [vcstools] Tarball download unpack failed: u"geometry_experimental-release-release-indigo-tf2_py-0.5.5-0 is not a subdirectory with contents in members [u'geometry2-release-release-indigo-tf2_py-0.5.5-0']"[/vcstools]
ERROR [vcstools] Tarball download unpack failed: u"geometry_experimental-release-release-indigo-tf2_msgs-0.5.5-0 is not a subdirectory with contents in members [u'geometry2-release-release-indigo-tf2_msgs-0.5.5-0']"[/vcstools]
ERROR [vcstools] Tarball download unpack failed: u"geometry_experimental-release-release-indigo-tf2-0.5.5-0 is not a subdirectory with contents in members [u'geometry2-release-release-indigo-tf2-0.5.5-0']"[/vcstools]
Exception caught during install: Error processing 'tf2' : [tf2] Checkout of https://github.com/ros-gbp/geometry_experimental-release/archive/release/indigo/tf2/0.5.5-0.tar.gz version geometry_experimental-release-release-indigo-tf2-0.5.5-0 into /bla/src/tf2 failed.
Error processing 'tf2_msgs' : [tf2_msgs] Checkout of https://github.com/ros-gbp/geometry_experimental-release/archive/release/indigo/tf2_msgs/0.5.5-0.tar.gz version geometry_experimental-release-release-indigo-tf2_msgs-0.5.5-0 into /bla/src/tf2_msgs failed.
Error processing 'tf2_py' : [tf2_py] Checkout of https://github.com/ros-gbp/geometry_experimental-release/archive/release/indigo/tf2_py/0.5.5-0.tar.gz version geometry_experimental-release-release-indigo-tf2_py-0.5.5-0 into /bla/src/tf2_py failed.
Error processing 'tf2_ros' : [tf2_ros] Checkout of https://github.com/ros-gbp/geometry_experimental-release/archive/release/indigo/tf2_ros/0.5.5-0.tar.gz version geometry_experimental-release-release-indigo-tf2_ros-0.5.5-0 into /bla/src/tf2_ros failed.

ERROR in config: Error processing 'tf2' : [tf2] Checkout of https://github.com/ros-gbp/geometry_experimental-release/archive/release/indigo/tf2/0.5.5-0.tar.gz version geometry_experimental-release-release-indigo-tf2-0.5.5-0 into /bla/src/tf2 failed.
Error processing 'tf2_msgs' : [tf2_msgs] Checkout of https://github.com/ros-gbp/geometry_experimental-release/archive/release/indigo/tf2_msgs/0.5.5-0.tar.gz version geometry_experimental-release-release-indigo-tf2_msgs-0.5.5-0 into /bla/src/tf2_msgs failed.
Error processing 'tf2_py' : [tf2_py] Checkout of https://github.com/ros-gbp/geometry_experimental-release/archive/release/indigo/tf2_py/0.5.5-0.tar.gz version geometry_experimental-release-release-indigo-tf2_py-0.5.5-0 into /bla/src/tf2_py failed.
Error processing 'tf2_ros' : [tf2_ros] Checkout of https://github.com/ros-gbp/geometry_experimental-release/archive/release/indigo/tf2_ros/0.5.5-0.tar.gz version geometry_experimental-release-release-indigo-tf2_ros-0.5.5-0 into /bla/src/tf2_ros failed.
```

gavanderhoorn, 2018-07-05:
Yes. Manual patching of the `.rosinstall` file is needed in this case.

I'll push an updated version of the `.rosinstall` file and close this issue.

gavanderhoorn, 2018-07-05:
@git-afsantos: the updated `.rosinstall` file should make this work for you again.

gavanderhoorn, 2018-07-05:
Thanks for the `docker` command line, that was very convenient to reproduce the error.

commit ef4b89c3 (2018-07-05): 5ee65b0: update repository ref in rosinstall file. Fix #63.
closed 2018-07-05 by gavanderhoorn