robust-rosin/robust #64: Suggestion: please rebase instead of merge

gavanderhoorn, 2018-07-05:
The commit history shows a lot of "merged branch X of Y".

Could I sugges to use `git rebase` instead? Updates to bug descriptions and bugzoo manifests seem fairly orthogonal, so not maintaining the historical order of commits doesn't seem to important.

Rebasing keeps the history a bit saner imo.

wasowski, 2018-07-05:
will try (less experience with rebasing here but no problem)

gavanderhoorn, 2018-07-05:
Unless you've touched someone else's files, a `git fetch && git rebase origin/master` should not require any manual intervention. Then a `git push` should be clean, no merges needed.

gavanderhoorn, 2018-07-05:
Yes, branches are also fine. But before merging, you could still rebase, as that would make the merge of the branch a lot cleaner.