robust-rosin/robust #67: Build image for fc854e0

wasowski, 2018-07-05:
Another one that needs traveling further back.  The bug is in robust/geometry2/fc854e0 on the fc854e0 branch.  It is November 2013.

gavanderhoorn, 2018-11-25:
According to the [release notes](https://github.com/ros/geometry2/commit/e546946d8be7d5990ba0170919d717678de1f2a4#diff-c6902b515f73bb7559aae72c96f58c96) for the version of `tf2_tools` that got released after the PR was merged that introduced the changes described by this `.bug`, the issue that this PR was for is [ros/geometry_experimental#44](https://github.com/ros/geometry_experimental/issues/44).

However, following that link just leads to the PR again.

Not sure what is going on there.

gavanderhoorn, 2018-11-25:
Created `.rosinstall` file with current `master` of time machine.

Command line:

```
rosinstall_generator_tm.sh "2013-11-06T18:43:24Z" hydro --deps --deps-only --tar tf2_tools > deps.rosinstall
```

timestamp retrieved using:

```
get_issue_creation_date.py https://github.com/ros/geometry2/pull/44
```

Note: this is the creation date of the Pull Request that contains the fix.

The `.bug` does not refer to any issue that could be used.

gavanderhoorn, 2019-06-05:
#297.