robust-rosin/robust #69: tar extraction fails for renamed repositories

gavanderhoorn, 2018-07-05:
From https://github.com/robust-rosin/robust/issues/63#issuecomment-402677897:

> `geometry2` was called `geometry_experimental` at the time bug `5ee65b0` was reported. So the time machine generates a `.rosinstall` file that references `geometry_experimental`. In ros/geometry2#160, the repository was renamed to `geometry2`, but that doesn't change the old releases.
> 
> The tarball contains a directory called: `geometry2-release-release-indigo-tf2-0.5.5-0`, but the expected name is `geometry_experimental-release-release-indigo-tf2-0.5.5-0`. As `wstool` (via `vcstools`) checks that the value specified in the `version` field of the `.rosinstall` file actually exists as a directory in the tarball, it notices the missing directory and fails the entire operation.
> 
> Manually updating the `version` fields to read `geometry2-release-release-indigo-tf2-0.5.5-0` (for `tf2` fi) makes everything work again.

Tracking this in a separate issue to give it more visibility.

Not sure this is something we can automatically fix (or even detect), but at least we have a manual work-around.

gavanderhoorn, 2018-07-05:
Example of `wstool` errors seen when this happens:

```
[tf2] Fetching https://github.com/ros-gbp/geometry_experimental-release/archive/release/indigo/tf2/0.5.5-0.tar.gz (version geometry_experimental-release-release-indigo-tf2-0.5.5-0) to /bla/src/tf2
ERROR [vcstools] Tarball download unpack failed: u"geometry_experimental-release-release-indigo-tf2-0.5.5-0 is not a subdirectory with contents in members [u'geometry2-release-release-indigo-tf2-0.5.5-0']"[/vcstools]
Exception caught during install: Error processing 'tf2' : [tf2] Checkout of https://github.com/ros-gbp/geometry_experimental-release/archive/release/indigo/tf2/0.5.5-0.tar.gz version geometry_experimental-release-release-indigo-tf2-0.5.5-0 into /bla/src/tf2 failed.

ERROR in config: Error processing 'tf2' : [tf2] Checkout of https://github.com/ros-gbp/geometry_experimental-release/archive/release/indigo/tf2/0.5.5-0.tar.gz version geometry_experimental-release-release-indigo-tf2-0.5.5-0 into /bla/src/tf2 failed.
```

gavanderhoorn, 2018-07-05:
For TF2 specifically (ie: `geometry_experimental` and `geometry2`), this was reported on ROS Answers as well: [Errors installing rosdep](https://answers.ros.org/question/55363). See the second answer.

wasowski, 2018-07-05:
Created a soft reference from the paper repo issues: https://bitbucket.org/itu-square/2017-42-bugs-in-ros/issues/15/an-interesting-case-for-the-reproduction . I tagged @brabrand there and also here to maximize the chance that he gets a notification.

gavanderhoorn, 2018-07-06:
I reported this over at https://github.com/ros/geometry2/issues/306.

Directory names are generated by github based on repository name, so perhaps this *can* be corrected automatically.

wasowski, 2018-10-11:
Oh good that I have a cut off date for studying geometry2 issues.  Otherwise this project literally risked an infinite loop!

gavanderhoorn, 2019-03-05:
This was worked-around in #173 (script updates entries in `.rosinstall` file with old repository name).

closed 2019-03-05 by gavanderhoorn