robust-rosin/robust #77: Build image for geometry2/932c6ca

wasowski, 2018-07-05:
This one is still from the unsupported period (Jan 2014).

gavanderhoorn, 2018-11-25:
Created `.rosinstall` file with current `master` of time machine.

Command line:

```
rosinstall_generator_tm.sh "2014-01-29T17:46:25Z" hydro --deps --deps-only --tar tf2_ros > deps.rosinstall
```

timestamp retrieved using:

```
get_issue_creation_date.py https://github.com/ros/geometry2/pull/51
```

Note: this is the creation date of the Pull Request that contains the fix.

The `.bug` does not refer to any issue that could be used.

gavanderhoorn, 2019-03-06:
Question: the PR was submitted when Hydro was the most recent release. The fix was actually merged into Indigo by tfoote.

Should we use `hydro` or `indigo` here as the distro?

Problem is: at `2014-01-29T17:46:25Z` Indigo didn't exist yet as a release.

gavanderhoorn, 2019-03-06:
@wasowski ?

gavanderhoorn, 2019-05-29:
@wasowski: I'll need some input for this one. See my previous comment.

wasowski, 2020-02-19:
@gavanderhoorn Hydro seems a more reasonable choice than indigo. I will merge #257 so that the bug description is on the master branch. But hopefully this discussion will help you to remember that this is not finished.