robust-rosin/robust #79: Failed to build docker image for b166c93

git-afsantos, 2018-07-05:
I could build the image initially. As soon as I changed the buggy and fixed commits to ones of the development branches and added the initial test case to the repository, it stopped working. The build fails when building the package.

```
CMake Error at repo-under-test/kobuki_driver/CMakeLists.txt:60 (find_package):
  By not providing "Findrostest.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "rostest", but
  CMake did not find one.

  Could not find a package configuration file provided by "rostest" with any
  of the following names:

    rostestConfig.cmake
    rostest-config.cmake

  Add the installation prefix of "rostest" to CMAKE_PREFIX_PATH or set
  "rostest_DIR" to a directory containing one of the above files.  If
  "rostest" provides a separate development package or SDK, be sure it has
  been installed.

[...]
Invoking "cmake" failed
The command '/bin/sh -c echo "[ROBUST] attempting to build PUT..."  && echo "[ROBUST] is a build failure expected? ${IS_BUILD_FAILURE}."  && ./build.sh || [ "${IS_BUILD_FAILURE}" = "yes" ]' returned a non-zero code: 1
```

This is the command I used to build the Docker image.

```bash
docker build \
-t robustrosin/test-b166c93 \
-f ../../Dockerfile \
--build-arg UBUNTU_VERSION=trusty \
--build-arg IS_BUILD_FAILURE=no \
--build-arg USE_APT_OLD_RELEASES=no \
--build-arg ROS_DISTRO=indigo \
--build-arg CATKIN_PKG=kobuki_driver \
--build-arg REPO_FORK_URL=https://github.com/robust-rosin/kobuki_core \
--build-arg REPO_BUG_COMMIT=c054e5437f2662774e5677c75e17388c8b1f4206 \
--build-arg REPO_FIX_COMMIT=51e46da5e9cb075f6ffea60132178c8ef7807d05 \
.
```

git-afsantos, 2018-07-06:
I just noticed this is the same error (missing `rostest`) as in #20.

I am starting to suspect that it will be a recurring issue for C++ bugs.

git-afsantos, 2018-07-06:
I tried to add `rostest` manually to the `.rosinstall` file.

```yaml
- tar:
    local-name: rostest
    uri: https://github.com/ros-gbp/ros_comm-release/archive/release/indigo/rostest/1.11.19-0.tar.gz
    version: ros_comm-release-release-indigo-rostest-1.11.19-0
```

I am now facing the following error.

```
E: Unable to locate package ros-indigo-roslaunch
ERROR: the following rosdeps failed to install
  apt: command [apt-get install -y ros-indigo-roslaunch] failed
executing command [apt-get install -y ros-indigo-roslaunch]
The command '/bin/sh -c apt-get clean  && apt-get update  && rosdep init  && rosdep update  && rosdep install --from-paths src -i --rosdistro=${ROS_DISTRO} -y       --skip-keys="python-rosdep python-catkin-pkg python-rospkg"  && apt-get clean  && rm -rf /var/lib/apt/lists/*' returned a non-zero code: 1
```

gavanderhoorn, 2018-07-06:
hm. `rostest` has its own requirements.

What you could perhaps do, is ask the TM to create a `.rosinstall` file for you for the `rostest` pkg. So instead of `kobuki_driver`, specify `rostest`.

The resulting `.rosinstall` file should then contain the dependencies of `rostest`. Merging that with your current `.rosinstall` file (manually, in a text editor, or smarter, if you know/can find a way) could work around this.

git-afsantos, 2018-07-06:
> What you could perhaps do, is ask the TM to create a `.rosinstall` file for you for the `rostest` pkg.

This seems to have worked. The build took much longer, but at least it finished without errors. I just need to run a couple tests, to see if everything is alright.

gavanderhoorn, 2018-07-06:
It was just a test, not something we should probably do for all the images.

git-afsantos, 2018-07-06:
It seems that everything is alright with this image, after manually adding `rostest`. So now we just need a solution for this issue in general.

closed 2018-07-06 by git-afsantos