robust-rosin/robust #80: Missing dependency on rostest

git-afsantos, 2018-07-06:
This has happened more than once now, when building Docker images for C++ bugs, after test development has started (#20, #79).

Developing C++ tests introduces a dependency on `rostest`, which the PUT originally did not have. As such, the generated `.rosinstall` files do not include `rostest`, and trying to build the images after test development has started (and the target commit hashes changed) fails.

We have to come up with a way of making sure the images have `rostest` available.

@gavanderhoorn @ChrisTimperley

gavanderhoorn, 2018-07-06:
Fixing https://github.com/rosin-project/rosinstall_generator_time_machine/issues/4 probably would make this one easier to address as well: we could then simply include `rostest` as an additional package in the call to `rosinstall_generator_tm`.

wasowski, 2018-10-12:
Where are we on this? (It seems that we need a historical version of rostest).