robust-rosin/robust #82: build.sh fails for b166c93

git-afsantos, 2018-07-06:
This is the last issue I have to get this bug done.

After I run `fix`, I try to run `build.sh` to compile the fixed code, but this fails with the following output.

```
root@f0b342ac1dc7:/ros_ws# ./build.sh 
Base path: /ros_ws
Source space: /ros_ws/src
Build space: /ros_ws/build
Devel space: /ros_ws/devel
Install space: /ros_ws/install
Whitelisted packages: kobuki_driver
####
#### Running command: "make cmake_check_build_system" in "/ros_ws/build"
####
####
#### Running command: "make -j1 -l1" in "/ros_ws/build"
####
Scanning dependencies of target kobuki
[  7%] Building CXX object repo-under-test/kobuki_driver/src/driver/CMakeFiles/kobuki.dir/kobuki.cpp.o
make[2]: *** No rule to make target `/ros_ws/src/repo-under-test/kobuki_driver/build/version_info.cpp', needed by `repo-under-test/kobuki_driver/src/driver/CMakeFiles/kobuki.dir/__/__/build/version_info.cpp.o'.  Stop.
make[1]: *** [repo-under-test/kobuki_driver/src/driver/CMakeFiles/kobuki.dir/all] Error 2
make: *** [all] Error 2
Invoking "make -j1 -l1" failed
```

However, simply calling `catkin_make` works fine.

git-afsantos, 2018-11-24:
I will have to build this again, as my virtual machine ran out of space and I deleted a bunch of Docker images.

gavanderhoorn, 2018-11-25:
> I will have to build this again, as my virtual machine ran out of space and I deleted a bunch of Docker images.

*virtual machine*? :)

git-afsantos, 2018-11-25:
Yup. I have been doing most of this work within a VM in a Windows laptop. I should probably just go with dual boot, but this setup has worked so far.

I think either the issue got worse, or it is more intricate than what I first reported.

In a fresh container, running `unfix` and then `build.sh` results in an error.

> Whitelisted packages: kobuki_driver
>
>Running command: "make cmake_check_build_system" in "/ros_ws/build"
>
>
>Running command: "make -j1 -l1" in "/ros_ws/build"
>
> Scanning dependencies of target kobuki
make[2]: *** No rule to make target `/ros_ws/src/repo-under-test/kobuki_driver/build/version_info.cpp', needed by `repo-under-test/kobuki_driver/src/driver/CMakeFiles/kobuki.dir/__/__/build/version_info.cpp.o'.  Stop.
make[1]: *** [repo-under-test/kobuki_driver/src/driver/CMakeFiles/kobuki.dir/all] Error 2
make: *** [all] Error 2
**Invoking "make -j1 -l1" failed**

Then I ran `fix` and `build.sh`, which results in the same error above.

Afterwards, I ran `unfix` again, and `catkin_make`, which built successfully, and I was even able to run `test.sh`. At this time, `test.sh` built the test code.

Running `catkin_make` after `fix` *also* resulted in the error above this time.

Funny thing is, I ran `unfix` and `build.sh` again, but this time it built. Presumably, the initial `catkin_make` handled the issue.

Funnier yet, I finally ran `fix` and `catkin_make`, just after the successful `catkin_make` from `unfix`, *and now it built*. I was also able to run `test.sh` in the fixed version this time.

git-afsantos, 2018-11-25:
Just reproduced the exact same sequence of steps in another container, and the results are the same. At least, the issue seems to be deterministic.

In the last step, `fix` and `build.sh` results in an error, but `fix` and `catkin_make` works.