robust-rosin/robust #87: What is the bug in geomtry2/pr60

wasowski, 2018-08-03:
There is very little to go by. 
The commit: https://github.com/ros/geometry2/pull/60/commits/edf582c12b6bfdf345181ed8cca776918d70a7dd
and the PR: https://github.com/ros/geometry2/pull/60

My C++ is rusty. This looks like some murky corner, a gcc bug, or a more recent (as in this century) change to C++ rules.  Anybody knows?  If we cannot understand if this is a gcc bug, or a bug in rosin, I will have to skip this one.  Comments?

@gavanderhoorn @ChrisTimperley @ipa-jfh @git-afsantos

gavanderhoorn, 2018-08-03:
I'm not sure.

The [changelog](https://github.com/ros/geometry2/commit/11fd4ba81553600f43a2c926a49243f935d88315#diff-fac65ee6a2cb7cb70c817a4467e96a20) also doesn't help too much.

As `this->` was added it could be that GCC4.9 now requires this or without `this->` the invocation was ambiguous.

wasowski, 2018-08-03:
I did google a lot, and found no obvious thing, neither with the `this` pointer (hard to google for) and with gcc 4.9.  (for example, this https://www.geeksforgeeks.org/this-pointer-in-c/ site, still reports rules like in the old days).

The `signalMessage` function seems to be inherited from `SimpleFilter` (https://github.com/ros/ros_comm/blob/e96c407c64e1c17b0dd2bb85b67f388380527097/utilities/message_filters/include/message_filters/simple_filter.h). The method is overloaded but adding a `this` qualification would not disambiguate this overloading, I think.  I don't see a local `signalMessage` in `MessageFilter` (`this` would not help either), and I don't see a local variable with the same name.  Something more subtle must be going on (and something that only started to matter with gcc 4.9 - an ambiguity would likely be a problem before). 

It might be the case where we want to build the image before we can describe the bug ... It should be simply a build error - when we read the message, things will become clearer?