robust-rosin/robust #99: Update bug description files to use YAML file format

ChrisTimperley, 2018-09-19:


ChrisTimperley, 2018-09-19:
@wasowski @gavanderhoorn, any objections to this? It might be possible to (mostly) automate the construction of the `manifest.bugzoo.yml` files if the bug description files are written in YAML.

ChrisTimperley, 2018-09-19:
That would be great. I was thinking that we could put together a simple Python-based linter to check that all of the bug descriptions are well formed and complete (we could also add it to the CI). We should also be able to add all of the information needed by the time machine to the YAML file too? That way we could write a wrapper that calls the time machine using the information inside a given bug description file.

ChrisTimperley, 2018-09-19:
It looks like we could use [Rx](http://rx.codesimply.com/) to perform the validation: https://blog.picnic.nl/how-to-use-yaml-schema-to-validate-your-yaml-files-c82c049c2097

gavanderhoorn, 2018-09-19:
I like the Rx idea.

> We should also be able to add all of the information needed by the time machine to the YAML file too? That way we could write a wrapper that calls the time machine using the information inside a given bug description file.

yes, that would be nice. We'd have to see whether that would work for all cases though, as there can sometimes be quite a bit of ambiguity about which repository at which commit is ultimately involved and/or should be used for the `rosdistro` timestamp.

ChrisTimperley, 2018-09-19:
How do you feel about the `time-machine` section in https://github.com/robust-rosin/robust/blob/yaml-bug-descriptions/mavros/08cd181.bug?

gavanderhoorn, 2018-09-19:
The yaml that is generated right now is based on the dictionary that @wasowski and @brabrand proposed. The `reproduction-images` was specific terminology I believe. Perhaps we should keep that?

@wasowski: opinion?

Can we also please change `distro` to `ros_distro`? `distro` could also be "Linux distribution" or any other software distribution.

And `package` needs to be allowed to be a list for #56.

Would we want to make the name a bit more specific? `package` is rather generic.

ChrisTimperley, 2018-09-19:
I've put together a prototype script to generate ROS install files using the extra information provided by `time-machine` in the `.bug` files: https://github.com/robust-rosin/robust/blob/yaml-bug-descriptions/scripts/build-rosinstall.py

It needs a flag to toggle whether existing files should be overwritten, but it should work for `mavros/08cd181.bug`.

```
$ git submodule update --init --recursive
$ python scripts/build-rosinstall.py mavros/08cd181.bug
```

There's also an option to build all of the ROS install files for all `.bug` files in a given directory.

PR #100 (2018-10-01): Converted bug descriptions to YAML format (fixes #99)
Fixes #99
commit abce6859 (2018-10-01): Converted bug descriptions to YAML format (fixes #99) (#100)

* added manifest for 08cd181

* added manifest for 0e2ea0c

* updated one mavros bug to use YAML

* ported all mavros files

* remove old bugzoo files

* removed L1 and L3 examples

* added time-machine section to a mavros bug file

* added skeleton for build-rosinstall

* renamed distro to ros_distro and updated package to ros_pkgs

* added time_machine as a submodule

* undo borked git submodule

* removed borked time_machine

* fixed time_machine submodule

* moved time_machine to appropriate location

* fixed path issues

* adde [... cut]