robust-rosin/robust #101: Inconsistent use of `fix.fix-in` property in bug description files

ChrisTimperley, 2018-09-21:
The majority of bug description files use `fix.fix-in` to provide a list of the files that were modified by the fixing commit/PR, but a number use the property to provide the URL for the issue or pull request, such as `care-o-bot/43705f7.bug`.

gavanderhoorn, 2018-09-24:
With the diff, does it really help to know which files are affected by a fix?

ChrisTimperley, 2018-09-24:
I think it's convenient but not essential, especially for bugs that are fixed over the course of multiple commits.

wasowski, 2018-11-26:
I think the intention was to list the files. It is not the field that we intended to use a lot (it is redundant with the diff), but it does seem to give the reader a quick impression of what is going on - one learns a bit from the nature of the file names.  One can also query bugs by names of files they touch.

I am more inclined to keep the field than to remove it (but this is indeed secondary).  But if we go with this we should fix the inconsistent use (or at least file issues if we such - although it is probably faster to fix than to report issues).

ChrisTimperley, 2018-11-26:
> I think the intention was to list the files. It is not the field that we intended to use a lot (it is redundant with the diff), but it does seem to give the reader a quick impression of what is [...]

This can be quite useful when writing tests.