robust-rosin/robust #102: Inconsistent use of `bug.languages` and `fix.languages` in bug description files

ChrisTimperley, 2018-09-21:
A number of bug description files do not specify any languages in `bug.languages`, but give a list of the languages of the files that were edited in `fix.languages`. What should languages should `bug.languages` contain and why would that list of languages ever be different to `fix.languages`?

git-afsantos, 2018-09-21:
I think that decision had to do mostly with cases where the fix of the bug is not exactly where you perceive the bug to manifest (although this may be subjective).

Example: Python script fails to import a dependency
Fix: add missing dependency to package.xml
Bug language: Python
Fix language: package.xml

In this case you could argue that the bug is in the package.xml (missing dependency), so I am not sure whether we truly need to distinguish both cases.

gavanderhoorn, 2018-09-24:
I agree with @git-afsantos: (un)fortunately, bugs in ROS typically involve multiple languages.

So being able to express that in the manifests would be good.

wasowski, 2018-11-26:
Yes this is all according to the intention.  Bug is really the description of the failure.  So if a runtime of python reports a bug in l. 42 then we report python.  The bug could well be fixed in C++, then in fix.languages we would have C++ (or whatever it was fixed in). 

Sorry for missing this for so long.  Only now I realized I have such a backlog on this issue tracker.  Chris, I will close this one, but feel free to reopen if this is unclear. Also there is a doc that should be clarifying that - I hope it is not entirely hopeless.

closed 2018-11-26 by wasowski