robust-rosin/robust #103: Docker build fails for Kobuki f95c384

git-afsantos, 2018-09-21:
Tried to build with the following command.

```bash
$ docker build -t robustrosin/test-f95c384 -f ../../Dockerfile --build-arg UBUNTU_VERSION=precise --build-arg IS_BUILD_FAILURE=yes --build-arg USE_APT_OLD_RELEASES=no --build-arg ROS_DISTRO=hydro --build-arg CATKIN_PKG=kobuki_node --build-arg REPO_FORK_URL=https://github.com/robust-rosin/kobuki --build-arg REPO_BUG_COMMIT=446f0b0cf9699a69b4f2badfd45fee702cbf22bc --build-arg REPO_FIX_COMMIT=f95c3847d97fba46b6317e00ffd78ce352ec376f .
```

Error message (step 23/39):

> The command `'/bin/sh -c apt-get update  && apt-get install -y --no-install-recommends       build-essential       ca-certificates       git       python-pip       cmake       wget       lsb-release  && pip install --upgrade pip==9.0.3  && pip install setuptools  && pip install --upgrade       wheel       rosdep       wstool       rosinstall       rospkg       catkin_pkg  && apt-get clean  && rm -rf /var/lib/apt/lists/*'` returned a non-zero code: 1

Output right before the error:

> Downloading/unpacking pip==9.0.3
>   Cannot fetch index base URL http://pypi.python.org/simple/
>   Could not find any downloads that satisfy the requirement pip==9.0.3
> No distributions at all found for pip==9.0.3

Seems like something is missing. Shall I try something other than changing the distros?

The bug report is not very explicit regarding the distros they were using. It happened a few days after Indigo released, and they assigned the fix to the Indigo milestone, but the package had not been released yet in Indigo at the time of reporting (so the time machine cannot find the PUT).

I did some digging in the repository files (browsing at the buggy commit), and they seem to be using hydro, so I just assumed ROS Hydro + Ubuntu Precise.

git-afsantos, 2018-09-26:
@ChrisTimperley @gavanderhoorn

ChrisTimperley, 2018-10-01:
https://github.com/pypa/pip/issues/5156

closed 2018-10-01 by ChrisTimperley

gavanderhoorn, 2018-10-03:
This is a strange error: `yaml-cpp` is the `rosdep` *key*, not the `apt` package name (see [here](https://github.com/ros/rosdistro/blob/e0c861929155b2fce67376bc424fd11febb31852/rosdep/base.yaml#L5229-L5247)).

I'm not sure why `rosdep` is trying to install `yaml-cpp` directly.

commit ca492881 (2018-10-01): Fixed installation of pip for Ubuntu 12.04 (fixes #103) (#119)

* safely update pip to 9.0.3

* use specific ubuntu releases
commit f53ded2d (2018-10-01): Squashed commit of the following:

commit ca492881c79d2b3905bc6840431ae6020fb32a42
Author: Chris Timperley <christimperley@googlemail.com>
Date:   Mon Oct 1 19:36:04 2018 -0400

    Fixed installation of pip for Ubuntu 12.04 (fixes #103) (#119)

    * safely update pip to 9.0.3

    * use specific ubuntu releases