robust-rosin/robust #104: How to select the best PUT

git-afsantos, 2018-09-22:
I think it makes sense for the PUT to be whatever is in `fix.fix-in`, in the bug description, as it is where changes were made.

I am opening this issue just to gather an opinion on whether it is relevant for the `bug.package` packages to be considered the PUT.

Take Kobuki [e3c9bbc.bug](https://github.com/robust-rosin/robust/blob/master/kobuki/e3c9bbc.bug) as an example. The issue is that there was no `source` link in the rosdistro YAML. I assigned the missing Kobuki packages under `bug.package`, but the fix happens in `rosdistro`.

I can either use Kobuki's packages as PUT, and the test script gathers the rosdistro YAML somehow, or I can use rosdistro as the PUT, making sure the files are there and can easily be read/tested.

git-afsantos, 2018-11-24:
Should we remove `e3c9bbc` so as not to cause confusion, and since it is clearly an outlier? Or do we leave it precisely as an example of what not to reproduce?

gavanderhoorn, 2018-11-24:
I'm not sure. Leaving it makes as much sense to me as removing it.

I was not the one who added `e3c9bbc`, so perhaps @wasowski or @brabrand can clarify what the intention was?

git-afsantos, 2018-11-24:
Actually, I was the one who added it. I believe at the time we were not yet discussing bug reproduction, otherwise I would have skipped it. I think the goal at the time was to collect bugs as diverse as possible, but it is debatable whether this can be considered a bug.