robust-rosin/robust #105: Need support for preparing the container [105dc16.bug]

ipa-jfh, 2018-09-24:
@wasowski or @gavanderhoorn can you please give me some support:

I am trying to get started with the [105dc16.bug](https://github.com/robust-rosin/robust/blob/d50d51b/care-o-bot/105dc16.bug). 

In order to get the `.rosinstall` file I run in the venv the command `./rosinstall_generator_tm.sh https://github.com/ipa320/cob_control/issues/6  105dc16   indigo  cob_control deps_105dc16.rosinstall`. However, the generated file is empty, apparently due to:

> // Log
> The following not released packages/stacks will be ignored: cob_control
> No packages/stacks left after ignoring not released

> // From the README
> Make sure this is the case if running into this error. Realise that the package may not have been released at the time the bug was reported (if using the issue URL as input to the time machine).

I checked and indeed the repo is not released at the time of the bug. Is there currently a workaround for this?

---

Assuming the file is generated successfully, how do I add the bug to the Bugzoo list `bugzoo bug list`?

---

We still use branches of the form `bugid_robust_fixed` and/or corresponding git tags of the form `bugid_robust_fixed_released`?

gavanderhoorn, 2018-09-24:
> In order to get the `.rosinstall` file I run in the venv the command `./rosinstall_generator_tm.sh https://github.com/ipa320/cob_control/issues/6 105dc16 indigo cob_control deps_105dc16.rosinstall`. [...]

I feel we've had a case like this before, but cannot find the issue where we discussed it.

@git-afsantos: I seem to remember that it was one of the Kobuki bugs. Would you happen to remember?

git-afsantos, 2018-09-24:
Yes, we had a case like this before in #55 . The solution was not pretty, though. I will leave it here for convenience (and maybe a new proposal).

> For this bug, the problem was that the GitHub issue was opened before the PUT was added to rosdistro. This makes the time machine unable to produce all dependencies using the issue report date.
>
> The solution cannot be fully automated at the moment, and it consisted on two steps.
>
> 1. Deleting the cache and the bughunt_5ee65b0 branch from rosdistro repository used by the time machine.
>
> 2. Using an explicit date, instead of an issue URL as a script argument. This has to be a manually selected date, ideally right after the date when the PUT was added to rosdistro.
>
> The solution incurs a small risk of not using exactly the same dependency versions that were used when the bug was reported -- we are selecting a date slightly further into the future. In this particular case, the risk is very small, since it is only a 3 day difference.

gavanderhoorn, 2018-09-24:
@git-afsantos: yes, that was it. Thanks.

@ipa-jfh: I don't know of a way to automate this with the time machine. `rosinstall_generator` works with released packages, so if they're not released, I cannot ask it to generate a `.rosinstall` file.

---

Edit: hm, that's not entirely true actually. `rosinstall_generator` *can* work with source repositories, but I'd have to investigate this. If we run into more bugs which need to build PUTs from source this might be worth it.

gavanderhoorn, 2018-09-24:
Tracking (part of) this in #107.

ipa-jfh, 2018-09-25:
Thank you both for the information!

It compiled the list successfully with another date (roughly a month after the issue). Probably this bug would also "work" with that date.

How do I proceed with this `.rosinstall` file to get the Docker container (through BugZoo)?

gavanderhoorn, 2018-09-25:
You could take a look at #60 (rendered version: [here](https://github.com/robust-rosin/robust/blob/554fbb37803e17656ada29b8d57b5e4b8a1fcf6d/doc/creating-bugzoo-snapshot.md)). @ChrisTimperley is changing things though, so unfortunately that document might be a bit out-of-date.