robust-rosin/robust #120: mavros/1f01916 fails to build

ChrisTimperley, 2018-10-03:
```
Step 28/40 : RUN wstool init -j8 ${ROS_WSPACE}/src ${ROS_WSPACE}/deps.rosinstall

 ---> Running in 0afa02aea419
Using initial elements from: /ros_ws/deps.rosinstall
Writing /ros_ws/src/.rosinstall
[cpp_common] Fetching https://github.com/ros-gbp/roscpp_core-release/archive/release/hydro/cpp_common/0.4.3-0.tar.gz (version roscpp_core-release-release-hydro-cpp_common-0.4.3-0) to /ros_ws/src/cpp_common
[cpp_common] Done.
[class_loader] Fetching https://github.com/ros-gbp/class_loader-release/archive/release/hydro/class_loader/0.2.5-0.tar.gz (version class_loader-release-release-hydro-class_loader-0.2.5-0) to /ros_ws/src/class_loader
[class_loader] Done.
[angles] Fetching https://github.com/ros-gbp/geometry_angles_utils-release/archive/release/hydro/angles/1.9.9-0.tar.gz (version geometry_angles_utils-release-release-hydro-angles-1.9.9-0) to /ros_ws/src/angles
[angles] Done.
[actionlib] Fetching https://github.com/ros-gbp/actionlib-release/archive/release/hydro/actionlib/1.10.3-0.tar.gz (version actionlib-release-release-hydro-actionlib-1.10.3-0) to /ros_ws/src/actionlib
[actionlib] Done.
[catkin] Fetching https://github.com/ros-gbp/catkin-release/archive/release/hydro/catkin/0.5.88-0.tar.gz (version catkin-release-release-hydro-catkin-0.5.88-0) to /ros_ws/src/catkin
[catkin] Done.
[diagnostic_msgs] Fetching https://github.com/ros-gbp/common_msgs-release/archive/release/hydro/diagnostic_msgs/1.10.6-0.tar.gz (version common_msgs-release-release-hydro-diagnostic_msgs-1.10.6-0) to /ros_ws/src/diagnostic_msgs
[diagnostic_msgs] Done.
[console_bridge] Fetching https://github.com/ros-gbp/console_bridge-release/archive/release/hydro/console_bridge/0.2.4-1.tar.gz (version console_bridge-release-release-hydro-console_bridge-0.2.4-1) to /ros_ws/src/console_bridge
[console_bridge] Done.
[actionlib_msgs] Fetching https://github.com/ros-gbp/common_msgs-release/archive/release/hydro/actionlib_msgs/1.10.6-0.tar.gz (version common_msgs-release-release-hydro-actionlib_msgs-1.10.6-0) to /ros_ws/src/actionlib_msgs
[actionlib_msgs] Done.
[actionlib,actionlib_msgs,angles,catkin,class_loader,console_bridge,cpp_common,diagnostic_msgs] still active
[gencpp] Fetching https://github.com/ros-gbp/gencpp-release/archive/release/hydro/gencpp/0.4.17-0.tar.gz (version gencpp-release-release-hydro-gencpp-0.4.17-0) to /ros_ws/src/gencpp
[gencpp] Done.
[diagnostic_updater] Fetching https://github.com/ros-gbp/diagnostics-release/archive/release/hydro/diagnostic_updater/1.8.4-0.tar.gz (version diagnostics-release-release-hydro-diagnostic_updater-1.8.4-0) to /ros_ws/src/diagnostic_updater
[diagnostic_updater] Done.
[genmsg] Fetching https://github.com/ros-gbp/genmsg-release/archive/release/hydro/genmsg/0.4.25-0.tar.gz (version genmsg-release-release-hydro-genmsg-0.4.25-0) to /ros_ws/src/genmsg
[genmsg] Done.
[genpy] Fetching https://github.com/ros-gbp/genpy-release/archive/release/hydro/genpy/0.4.17-0.tar.gz (version genpy-release-release-hydro-genpy-0.4.17-0) to /ros_ws/src/genpy
[genpy] Done.
[geometry_msgs] Fetching https://github.com/ros-gbp/common_msgs-release/archive/release/hydro/geometry_msgs/1.10.6-0.tar.gz (version common_msgs-release-release-hydro-geometry_msgs-1.10.6-0) to /ros_ws/src/geometry_msgs
[geometry_msgs] Done.
[message_filters] Fetching https://github.com/ros-gbp/ros_comm-release/archive/release/hydro/message_filters/1.10.10-0.tar.gz (version ros_comm-release-release-hydro-message_filters-1.10.10-0) to /ros_ws/src/message_filters
[message_filters] Done.
[mavlink] Fetching https://github.com/vooon/mavlink-gbp-release/archive/release/hydro/mavlink/1.0.9-5.tar.gz (version mavlink-gbp-release-release-hydro-mavlink-1.0.9-5) to /ros_ws/src/mavlink
[mavlink] Done.
[genlisp] Fetching https://github.com/ros-gbp/genlisp-release/archive/release/hydro/genlisp/0.4.14-0.tar.gz (version genlisp-release-release-hydro-genlisp-0.4.14-0) to /ros_ws/src/genlisp
[genlisp] Done.
[diagnostic_updater,gencpp,genlisp,genmsg,genpy,geometry_msgs,mavlink,message_filters] still active
[message_generation] Fetching https://github.com/ros-gbp/message_generation-release/archive/release/hydro/message_generation/0.2.10-0.tar.gz (version message_generation-release-release-hydro-message_generation-0.2.10-0) to /ros_ws/src/message_generation
[message_generation] Done.
[message_runtime] Fetching https://github.com/ros-gbp/message_runtime-release/archive/release/hydro/message_runtime/0.4.12-0.tar.gz (version message_runtime-release-release-hydro-message_runtime-0.4.12-0) to /ros_ws/src/message_runtime
[message_runtime] Done.
[rosbuild] Fetching https://github.com/ros-gbp/ros-release/archive/release/hydro/rosbuild/1.10.10-0.tar.gz (version ros-release-release-hydro-rosbuild-1.10.10-0) to /ros_ws/src/rosbuild
[rosbuild] Done.
[pluginlib] Fetching https://github.com/ros-gbp/pluginlib-release/archive/release/hydro/pluginlib/1.9.24-0.tar.gz (version pluginlib-release-release-hydro-pluginlib-1.9.24-0) to /ros_ws/src/pluginlib
[pluginlib] Done.
[rosbag_storage] Fetching https://github.com/ros-gbp/ros_comm-release/archive/release/hydro/rosbag_storage/1.10.10-0.tar.gz (version ros_comm-release-release-hydro-rosbag_storage-1.10.10-0) to /ros_ws/src/rosbag_storage
[rosbag_storage] Done.
[rosconsole] Fetching https://github.com/ros-gbp/ros_comm-release/archive/release/hydro/rosconsole/1.10.10-0.tar.gz (version ros_comm-release-release-hydro-rosconsole-1.10.10-0) to /ros_ws/src/rosconsole
[rosconsole] Done.
[rosclean] Fetching https://github.com/ros-gbp/ros-release/archive/release/hydro/rosclean/1.10.10-0.tar.gz (version ros-release-release-hydro-rosclean-1.10.10-0) to /ros_ws/src/rosclean
[rosclean] Done.
[rosbag] Fetching https://github.com/ros-gbp/ros_comm-release/archive/release/hydro/rosbag/1.10.10-0.tar.gz (version ros_comm-release-release-hydro-rosbag-1.10.10-0) to /ros_ws/src/rosbag
[rosbag] Done.
[message_generation,message_runtime,pluginlib,rosbag,rosbag_storage,rosbuild,rosclean,rosconsole] still active
[roscpp_serialization] Fetching https://github.com/ros-gbp/roscpp_core-release/archive/release/hydro/roscpp_serialization/0.4.3-0.tar.gz (version roscpp_core-release-release-hydro-roscpp_serialization-0.4.3-0) to /ros_ws/src/roscpp_serialization
[roscpp_serialization] Done.
[roslang] Fetching https://github.com/ros-gbp/ros-release/archive/release/hydro/roslang/1.10.10-0.tar.gz (version ros-release-release-hydro-roslang-1.10.10-0) to /ros_ws/src/roslang
[roslang] Done.
[roslaunch] Fetching https://github.com/ros-gbp/ros_comm-release/archive/release/hydro/roslaunch/1.10.10-0.tar.gz (version ros_comm-release-release-hydro-roslaunch-1.10.10-0) to /ros_ws/src/roslaunch
[roslaunch] Done.
[roslib] Fetching https://github.com/ros-gbp/ros-release/archive/release/hydro/roslib/1.10.10-0.tar.gz (version ros-release-release-hydro-roslib-1.10.10-0) to /ros_ws/src/roslib
[roslib] Done.
[rosgraph] Fetching https://github.com/ros-gbp/ros_comm-release/archive/release/hydro/rosgraph/1.10.10-0.tar.gz (version ros_comm-release-release-hydro-rosgraph-1.10.10-0) to /ros_ws/src/rosgraph
[rosgraph] Done.
[rosgraph_msgs] Fetching https://github.com/ros-gbp/ros_comm-release/archive/release/hydro/rosgraph_msgs/1.10.10-0.tar.gz (version ros_comm-release-release-hydro-rosgraph_msgs-1.10.10-0) to /ros_ws/src/rosgraph_msgs
[rosgraph_msgs] Done.
[roscpp_traits] Fetching https://github.com/ros-gbp/roscpp_core-release/archive/release/hydro/roscpp_traits/0.4.3-0.tar.gz (version roscpp_core-release-release-hydro-roscpp_traits-0.4.3-0) to /ros_ws/src/roscpp_traits
[... 34 lines cut ...]
[rostime] Done.
[roswtf] Fetching https://github.com/ros-gbp/ros_comm-release/archive/release/hydro/roswtf/1.10.10-0.tar.gz (version ros_comm-release-release-hydro-roswtf-1.10.10-0) to /ros_ws/src/roswtf
[roswtf] Done.
[rostest,rostime,rostopic,rosunit,roswtf,sensor_msgs,std_msgs,tf] still active
ERROR [vcstools] Tarball download unpack failed: u"geometry_experimental-release-release-hydro-tf2_py-0.4.11-0 is not a subdirectory with contents in members [u'geometry2-release-release-hydro-tf2_py-0.4.11-0']"[/vcstools]

[tf2_py] Fetching https://github.com/ros-gbp/geometry_experimental-release/archive/release/hydro/tf2_py/0.4.11-0.tar.gz (version geometry_experimental-release-release-hydro-tf2_py-0.4.11-0) to /ros_ws/src/tf2_py
ERROR [vcstools] Tarball download unpack failed: u"geometry_experimental-release-release-hydro-tf2_ros-0.4.11-0 is not a subdirectory with contents in members [u'geometry2-release-release-hydro-tf2_ros-0.4.11-0']"[/vcstools]

[... cut ...]
Error processing 'tf2_py' : [tf2_py] Checkout of https://github.com/ros-gbp/geometry_experimental-release/archive/release/hydro/tf2_py/0.4.11-0.tar.gz version geometry_experimental-release-release-hydro-tf2_py-0.4.11-0 into /ros_ws/src/tf2_py failed.
Error processing 'tf2_ros' : [tf2_ros] Checkout of https://github.com/ros-gbp/geometry_experimental-release/archive/release/hydro/tf2_ros/0.4.11-0.tar.gz version geometry_experimental-release-release-hydro-tf2_ros-0.4.11-0 into /ros_ws/src/tf2_ros failed.
```

ChrisTimperley, 2018-10-03:
Same issue occurs for `mavros/215010d`. Likely due to an incorrect rosinstall file.

gavanderhoorn, 2018-10-03:
Seeing as this is TF2, I'm guessing this is #69.

gavanderhoorn, 2018-10-03:
It's not really incorrect.

The repository was renamed. It's an unfortunate combination of how github generates tarballs and `rosdep` checks for "successful" extraction.

gavanderhoorn, 2018-10-03:
Similar to #117.

ChrisTimperley, 2018-10-30:
Manually fixed via: `%s/geometry_experimental/geometry2/g`

commit 61c0a89f (2018-10-09): manually fixed #120 via sed
closed 2018-10-30 by ChrisTimperley