robust-rosin/robust #122: Kobuki 17560e9 builds but does not reproduce

git-afsantos, 2018-10-03:
This is a build bug. The build is supposed to complain about missing dependencies in the buggy commit, but it does not happen for me in the container.

Within the container I ran

```bash
$ ./unfix
$ ./build.sh
```

I also tried with

```bash
$ ./unfix
$ catkin_make --only-pkg-with-deps=yocs_navigator --force-cmake
```

Here is the description of the bug and the [original issue](https://github.com/yujinrobot/kobuki/issues/348), so you do not have to look for it.

> When building the `yocs_navigator` package, a user experienced a C++ compilation error, stating that one of the header files was missing. This header file declares some custom ROS messages.
> [...]
> In this case, a dependency was missing for the faulty build target.  The dependency generated custom ROS messages, among which was the missing file.
> In the end, it is possible that the file was not missing, but actually being built after the target (due to the missing dependency), causing the error.

Could it be that catkin/cmake are building things in the right order, by chance, thus hiding the bug?

gavanderhoorn, 2018-10-03:
hm.

`catkin_make` should definitely fail here, as it tries to parallelise as much as possible the build.

As the `add_dependency(..)` is missing, it will not realise that the generated code for msgs is a requirement for the `yocs_basic_move_controller` target and that results in the *No such file or directory* error.

`catkin_tools` works differently so could hide this, yes.

But you already used `catkin_make` as well. That should fail.

Which commit is used as the 'buggy' commit? https://github.com/yujinrobot/yujin_ocs/commit/ecb7a6619b8c541b5c6042d3137969302604ca84?

git-afsantos, 2018-10-03:
Indeed. I ran into similar bugs myself, when trying to compile code from a project from our partners, so I expected this to be straightforward.

Yes, I picked that commit as the buggy. Blindly going for the parent commit has resulted so far, so I did not give it much thought. Do you think it might be the cause?