robust-rosin/robust #125: PUT fails to build for mavros/08cd181

ChrisTimperley, 2018-10-09:
```
/ros_ws/src/repo-under-test/libmavconn/src/interface.cpp: In function ‘mavconn::MAVConnInterface::Ptr mavconn::url_parse_serial(std::string, std::string, uint8_t, uint8_t)’:
/ros_ws/src/repo-under-test/libmavconn/src/interface.cpp:183:9: error: ‘make_shared’ is not a member of ‘boost’
/ros_ws/src/repo-under-test/libmavconn/src/interface.cpp:183:9: note: suggested alternative:
/usr/include/c++/4.6/bits/shared_ptr.h:547:5: note:   ‘std::make_shared’
/ros_ws/src/repo-under-test/libmavconn/src/interface.cpp:183:41: error: expected primary-expression before ‘>’ token

/ros_ws/src/repo-under-test/libmavconn/src/interface.cpp: In function ‘mavconn::MAVConnInterface::Ptr mavconn::url_parse_udp(std::string, std::string, uint8_t, uint8_t)’:
/ros_ws/src/repo-under-test/libmavconn/src/interface.cpp:209:9: error: ‘make_shared’ is not a member of ‘boost’
/ros_ws/src/repo-under-test/libmavconn/src/interface.cpp:209:9: note: suggested alternative:
/usr/include/c++/4.6/bits/shared_ptr.h:547:5: note:   ‘std::make_shared’
/ros_ws/src/repo-under-test/libmavconn/src/interface.cpp:209:38: error: expected primary-expression before ‘>’ token

/ros_ws/src/repo-under-test/libmavconn/src/interface.cpp: In function ‘mavconn::MAVConnInterface::Ptr mavconn::url_parse_tcp_client(std::string, std::string, uint8_t, uint8_t)’:
/ros_ws/src/repo-under-test/libmavconn/src/interface.cpp:225:9: error: ‘make_shared’ is not a member of ‘boost’
/ros_ws/src/repo-under-test/libmavconn/src/interface.cpp:225:9: note: suggested alternative:
/usr/include/c++/4.6/bits/shared_ptr.h:547:5: note:   ‘std::make_shared’
/ros_ws/src/repo-under-test/libmavconn/src/interface.cpp:225:44: error: expected primary-expression before ‘>’ token

/ros_ws/src/repo-under-test/libmavconn/src/interface.cpp: In function ‘mavconn::MAVConnInterface::Ptr mavconn::url_parse_tcp_server(std::string, std::string, uint8_t, uint8_t)’:
/ros_ws/src/repo-under-test/libmavconn/src/interface.cpp:240:9: error: ‘make_shared’ is not a member of ‘boost’
/ros_ws/src/repo-under-test/libmavconn/src/interface.cpp:240:9: note: suggested alternative:
/usr/include/c++/4.6/bits/shared_ptr.h:547:5: note:   ‘std::make_shared’
/ros_ws/src/repo-under-test/libmavconn/src/interface.cpp:240:44: error: expected primary-expression before ‘>’ token

/ros_ws/src/repo-under-test/libmavconn/src/tcp.cpp: In member function ‘void mavconn::MAVConnTCPServer::do_accept()’:
/ros_ws/src/repo-under-test/libmavconn/src/tcp.cpp:327:20: error: ‘make_shared’ is not a member of ‘boost’
/ros_ws/src/repo-under-test/libmavconn/src/tcp.cpp:327:20: note: suggested alternative:
/usr/include/c++/4.6/bits/shared_ptr.h:547:5: note:   ‘std::make_shared’
/ros_ws/src/repo-under-test/libmavconn/src/tcp.cpp:327:55: error: expected primary-expression before ‘>’ token

make[2]: *** [repo-under-test/libmavconn/CMakeFiles/mavconn.dir/src/interface.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....

make[2]: *** [repo-under-test/libmavconn/CMakeFiles/mavconn.dir/src/tcp.cpp.o] Error 1

make[1]: *** [repo-under-test/libmavconn/CMakeFiles/mavconn.dir/all] Error 2

make: *** [all] Error 2
```

gavanderhoorn, 2018-10-10:
```gcc
/ros_ws/src/repo-under-test/libmavconn/src/interface.cpp:183:9: error: ‘make_shared’ is not a member of ‘boost’
/ros_ws/src/repo-under-test/libmavconn/src/interface.cpp:183:9: note: suggested alternative:
/usr/include/c++/4.6/bits/shared_ptr.h:547:5: note:   ‘std::make_shared’
```

are we building something with / without C++11 support while it should (not) be?

ChrisTimperley, 2018-10-23:
This looks like the expected behaviour. Here's the fix commit: https://github.com/mavlink/mavros/commit/08cd18164e729b0fb0ed1f6b553e5458eb9b6a4c

I'm trying to build with the fixed version now to confirm.

ChrisTimperley, 2018-10-23:
Confirmed.

closed 2018-10-23 by ChrisTimperley