robust-rosin/robust #126: PUT fails to build for mavros/248cb38

ChrisTimperley, 2018-10-09:
```
Step 39/41 : RUN echo "[ROBUST] attempting to build PUT..."  && echo "[ROBUST] is a build failure expected? ${IS_BUILD_FAILURE}."  && ./build.sh || [ "${IS_BUILD_FAILURE}" = "yes" ]

 ---> Running in bef622cb1207
[ROBUST] attempting to build PUT...
[ROBUST] is a build failure expected? no.
-- The C compiler identification is GNU 4.8.4
-- The CXX compiler identification is GNU 4.8.4
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Using CATKIN_DEVEL_PREFIX: /ros_ws/devel
-- Using CMAKE_PREFIX_PATH: /opt/ros/indigo
-- This workspace overlays: /opt/ros/indigo
-- Found PythonInterp: /usr/bin/python (found version "2.7.6")
-- Using PYTHON_EXECUTABLE: /usr/bin/python
-- Using Debian Python package layout
-- Using empy: /usr/bin/empy
-- Using CATKIN_ENABLE_TESTING: ON
-- Call enable_testing()
-- Using CATKIN_TEST_RESULTS_DIR: /ros_ws/build/test_results
-- Looking for include file pthread.h
-- Looking for include file pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE
-- Found gtest sources under '/usr/src/gtest': gtests will be built
-- Using Python nosetests: /usr/bin/nosetests-2.7
-- catkin 0.6.14
-- BUILD_SHARED_LIBS is on
-- Using CATKIN_WHITELIST_PACKAGES: libmavconn;mavros
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- ~~  traversing 2 packages in topological order:
-- ~~  - libmavconn
-- ~~  - mavros
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- +++ processing catkin package: 'libmavconn'
-- ==> add_subdirectory(repo-under-test/libmavconn)
-- Boost version: 1.54.0
-- Found the following Boost libraries:
--   system
-- Performing Test COMPILER_SUPPORTS_STD_CXX11
-- Performing Test COMPILER_SUPPORTS_STD_CXX11 - Success
-- Selected MAVLink dialect: ardupilotmega
-- +++ processing catkin package: 'mavros'
-- ==> add_subdirectory(repo-under-test/mavros)
-- Using these message generators: gencpp;genlisp;genpy
CMake Error at /opt/ros/indigo/share/catkin/cmake/catkinConfig.cmake:75 (find_package):
  Could not find a package configuration file provided by "eigen_conversions"
  with any of the following names:

    eigen_conversionsConfig.cmake
    eigen_conversions-config.cmake

  Add the installation prefix of "eigen_conversions" to CMAKE_PREFIX_PATH or
  set "eigen_conversions_DIR" to a directory containing one of the above
  files.  If "eigen_conversions" provides a separate development package or
  SDK, be sure it has been installed.
Call Stack (most recent call first):
  repo-under-test/mavros/CMakeLists.txt:7 (find_package)
```

gavanderhoorn, 2018-10-10:
This is typicaly output if a pkg didn't declare its dependencies properly.

If this is not supposed to be a `BUILD_FAILURE` we might need to see how we can extend the metadata for a bug so we can store a list of additional dependencies that should end up in the Docker image -- besides the ones from the `package.xml` files.

ChrisTimperley, 2018-10-10:
> This is typically seen if a pkg didn't declare its dependencies properly.
> 
> If this is not supposed to be a `BUILD_FAILURE` we might need to see how we can extend the metadata for a bug so we can store a list of additional dependencies that should end up in the Docker image -- besides the ones from the `package.xml` files.

Might it be best to tweak the time machine to take the names of additional dependencies/packages as part of its arguments? If so, it should be easy enough to add an extra field to the `time-machine` section of the `.bug` files that provides the necessary information.

gavanderhoorn, 2018-10-10:
I think I would have a problem with that conceptually: those pkgs would not be PUTs, but additional dependencies.

The extra meta-data we would need would be akin to a 'patch' to the dependency declarations of packages. Adding them as additional PUTs seems wrong to me.

ChrisTimperley, 2018-11-13:
This now builds using a `.rosinstall` file generated by the time machine using the following options:
```
./rosinstall_generator_tm.sh "2015-07-28T14:18:43-040" indigo mavros eigen_conversions --deps
```

closed 2018-11-13 by ChrisTimperley

gavanderhoorn, 2018-11-13:
Don't forget to add `--tar` :)