robust-rosin/robust #127: PUT fails to build for mavros/282c9be

ChrisTimperley, 2018-10-09:
```
Step 39/41 : RUN echo "[ROBUST] attempting to build PUT..."  && echo "[ROBUST] is a build failure expected? ${IS_BUILD_FAILURE}."  && ./build.sh || [ "${IS_BUILD_FAILURE}" = "yes" ]

 ---> Running in c1dbb6c21f40
[ROBUST] attempting to build PUT...
[ROBUST] is a build failure expected? yes.
-- The C compiler identification is GNU
-- The CXX compiler identification is GNU
-- Check for working C compiler: /usr/bin/gcc
-- Check for working C compiler: /usr/bin/gcc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Using CATKIN_DEVEL_PREFIX: /ros_ws/devel
-- Using CMAKE_PREFIX_PATH: /opt/ros/hydro
-- This workspace overlays: /opt/ros/hydro
-- Found PythonInterp: /usr/bin/python (found version "2.7.3")
-- Using PYTHON_EXECUTABLE: /usr/bin/python
-- Python version: 2.7
-- Using Debian Python package layout
-- Using CATKIN_ENABLE_TESTING: ON
-- Call enable_testing()
-- Using CATKIN_TEST_RESULTS_DIR: /ros_ws/build/test_results
-- Looking for include files CMAKE_HAVE_PTHREAD_H
-- Looking for include files CMAKE_HAVE_PTHREAD_H - found
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE
-- Found gtest sources under '/usr/src/gtest': gtests will be built
-- catkin 0.5.88
-- BUILD_SHARED_LIBS is on
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- ~~  traversing 1 packages in topological order:
-- ~~  - mavros
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- +++ processing catkin package: 'mavros'
-- ==> add_subdirectory(repo-under-test)
-- Using these message generators: gencpp;genlisp;genpy
-- Boost version: 1.46.1
-- Found the following Boost libraries:
--   system
--   thread
--   date_time
--   filesystem
-- Could NOT find libev (missing:  libev_LIBRARY libev_INCLUDE_DIR)
-- The libev package not found. please install libev-dev
-- mavros: 8 messages, 15 services
-- Performing Test COMPILER_SUPPORTS_CXX11
-- Performing Test COMPILER_SUPPORTS_CXX11 - Failed
-- Performing Test COMPILER_SUPPORTS_CXX0X
-- Performing Test COMPILER_SUPPORTS_CXX0X - Success
-- Configuring incomplete, errors occurred!
CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:
libev_INCLUDE_DIR (ADVANCED)
   used as include directory in directory /ros_ws/src/repo-under-test
libev_LIBRARY (ADVANCED)
    linked by target "mavconn" in directory /ros_ws/src/repo-under-test
```

gavanderhoorn, 2018-10-10:
Perhaps the same as #126.

ChrisTimperley, 2018-11-21:
It looks like `libev` is a missing binary dependency that *should* have been picked up during `rosdep install`. If I'm not mistaken, I think that we'll need to add a new argument to the Dockerfile that provides a list of missing binary dependencies, which will be passed onto `apt-get install -y`. Thoughts, @gavanderhoorn

gavanderhoorn, 2018-11-21:
I'm not entirely sure why `rosdep` doesn't install the dependency (we have a similar problem in one of the other bugs of @git-afsantos where `libusb-dev` is missing).

One thing I *could* see happening is if the missing dependency is actually a transitive one, and somewhere between now and when the bug was actually reported dependencies of the top-level package were cleaned up leading to the transitive dependency no longer getting installed.

As we're not reverting the `rosdep` database, that is a possibility.

In other words: the dependency that is now missing was getting installed as part of the dependencies of some other dependency. Now that transitive dependency has been broken, it no longer gets installed leading to the issue.