robust-rosin/robust #128: PUT fails to build for mavros/29af3a3

ChrisTimperley, 2018-10-09:
```
Step 39/41 : RUN echo "[ROBUST] attempting to build PUT..."  && echo "[ROBUST] is a build failure expected? ${IS_BUILD_FAILURE}."  && ./build.sh || [ "${IS_BUILD_FAILURE}" = "yes" ]

 ---> Running in 8434fb97b49f
[ROBUST] attempting to build PUT...
[ROBUST] is a build failure expected? no.
-- The C compiler identification is GNU 4.8.4
-- The CXX compiler identification is GNU 4.8.4
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Using CATKIN_DEVEL_PREFIX: /ros_ws/devel
-- Using CMAKE_PREFIX_PATH: /opt/ros/indigo
-- This workspace overlays: /opt/ros/indigo
-- Found PythonInterp: /usr/bin/python (found version "2.7.6")
-- Using PYTHON_EXECUTABLE: /usr/bin/python
-- Using Debian Python package layout
-- Using empy: /usr/bin/empy
-- Using CATKIN_ENABLE_TESTING: ON
-- Call enable_testing()
-- Using CATKIN_TEST_RESULTS_DIR: /ros_ws/build/test_results
-- Looking for include file pthread.h
-- Looking for include file pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE
-- Found gtest sources under '/usr/src/gtest': gtests will be built
-- Using Python nosetests: /usr/bin/nosetests-2.7
-- catkin 0.6.9
-- BUILD_SHARED_LIBS is on
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- ~~  traversing 3 packages in topological order:
-- ~~  - libmavconn
-- ~~  - mavros_msgs
-- ~~  - mavros
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- +++ processing catkin package: 'libmavconn'
-- ==> add_subdirectory(repo-under-test/libmavconn)
-- Boost version: 1.54.0
-- Found the following Boost libraries:
--   system
-- Performing Test COMPILER_SUPPORTS_STD_CXX11
-- Performing Test COMPILER_SUPPORTS_STD_CXX11 - Success
-- Selected MAVLink dialect: ardupilotmega
-- +++ processing catkin package: 'mavros_msgs'
-- ==> add_subdirectory(repo-under-test/mavros_msgs)
-- Using these message generators: gencpp;genlisp;genpy
-- mavros_msgs: 24 messages, 27 services
-- +++ processing catkin package: 'mavros'
-- ==> add_subdirectory(repo-under-test/mavros)
CMake Error at /opt/ros/indigo/share/catkin/cmake/catkinConfig.cmake:75 (find_package):
  Could not find a package configuration file provided by "nav_msgs" with any
  of the following names:

    nav_msgsConfig.cmake
    nav_msgs-config.cmake

  Add the installation prefix of "nav_msgs" to CMAKE_PREFIX_PATH or set
  "nav_msgs_DIR" to a directory containing one of the above files.  If
  "nav_msgs" provides a separate development package or SDK, be sure it has
  been installed.
Call Stack (most recent call first):
  repo-under-test/mavros/CMakeLists.txt:7 (find_package)

-- Configuring incomplete, errors occurred!
See also "/ros_ws/build/CMakeFiles/CMakeOutput.log".
See also "/ros_ws/build/CMakeFiles/CMakeError.log".
Invoking "cmake" failed
```

gavanderhoorn, 2018-10-10:
#126 again.

gavanderhoorn, 2019-03-05:
Removed the `time-machine` label as this is not a limitation of that script.

gavanderhoorn, 2019-03-05:
Ref: #254.