robust-rosin/robust #129: PUT fails to build for mavros/5353982

ChrisTimperley, 2018-10-10:
```
Building CXX object repo-under-test/mavros/CMakeFiles/mavros.dir/src/lib/uas_stringify.cpp.o
Building CXX object repo-under-test/mavros/CMakeFiles/mavros.dir/src/lib/uas_data.cpp.o
Building CXX object repo-under-test/mavros/CMakeFiles/mavros.dir/src/lib/uas_timesync.cpp.o
Building CXX object repo-under-test/mavros/CMakeFiles/mavros.dir/src/lib/uas_sensor_orientation.cpp.o
Building CXX object repo-under-test/mavros/CMakeFiles/mavros.dir/src/lib/uas_frame_conversions.cpp.o
Building CXX object repo-under-test/mavros/CMakeFiles/mavros.dir/src/lib/mavros.cpp.o
Building CXX object repo-under-test/mavros/CMakeFiles/mavros.dir/src/lib/uas_quaternion_utils.cpp.o
Building CXX object repo-under-test/mavros/CMakeFiles/mavros.dir/src/lib/mavlink_diag.cpp.o
Building CXX object repo-under-test/mavros/CMakeFiles/mavros.dir/src/lib/rosconsole_bridge.cpp.o
/ros_ws/src/repo-under-test/mavros/src/lib/uas_frame_conversions.cpp:50:18: error: macro "ROS_BREAK" passed 1 arguments, but takes just 0
   ROS_BREAK(false);
                  ^
/ros_ws/src/repo-under-test/mavros/src/lib/uas_frame_conversions.cpp:78:18: error: macro "ROS_BREAK" passed 1 arguments, but takes just 0
   ROS_BREAK(false);
                  ^

/ros_ws/src/repo-under-test/mavros/src/lib/uas_frame_conversions.cpp:133:18: error: macro "ROS_BREAK" passed 1 arguments, but takes just 0
   ROS_BREAK(false);
                  ^

/ros_ws/src/repo-under-test/mavros/src/lib/uas_frame_conversions.cpp:165:17: error: macro "ROS_BREAK" passed 1 arguments, but takes just 0
  ROS_BREAK(false);
                 ^

In file included from /ros_ws/src/repo-under-test/mavros/src/lib/mavros.cpp:18:0:
/opt/ros/indigo/include/mavlink/config.h:1:0: warning: "MAVLINK_VERSION" redefined [enabled by default]
 #define MAVLINK_VERSION "2016.1.8"
 ^
In file included from /opt/ros/indigo/include/mavlink/v1.0/ardupilotmega/ardupilotmega.h:656:0,
                 from /opt/ros/indigo/include/mavlink/v1.0/ardupilotmega/mavlink.h:25,
                 from /ros_ws/src/repo-under-test/libmavconn/include/mavconn/mavlink_dialect.h:57,
                 from /ros_ws/src/repo-under-test/libmavconn/include/mavconn/interface.h:38,
                 from /ros_ws/src/repo-under-test/mavros/include/mavros/mavros.h:23,
                 from /ros_ws/src/repo-under-test/mavros/src/lib/mavros.cpp:14:
/opt/ros/indigo/include/mavlink/v1.0/ardupilotmega/../common/common.h:729:0: note: this is the location of the previous definition
 #define MAVLINK_VERSION 3
 ^

/ros_ws/src/repo-under-test/mavros/src/lib/uas_frame_conversions.cpp: In function ‘const Affine3d& get_static_transform_affine(mavros::UAS::STATIC_TRANSFORM)’:
/ros_ws/src/repo-under-test/mavros/src/lib/uas_frame_conversions.cpp:50:3: error: ‘ROS_BREAK’ was not declared in this scope
   ROS_BREAK(false);
   ^

/ros_ws/src/repo-under-test/mavros/src/lib/uas_frame_conversions.cpp: In static member function ‘static Eigen::Quaterniond mavros::UAS::transform_orientation(const Quaterniond&, mavros::UAS::STATIC_TRANSFORM)’:
/ros_ws/src/repo-under-test/mavros/src/lib/uas_frame_conversions.cpp:78:3: error: ‘ROS_BREAK’ was not declared in this scope
   ROS_BREAK(false);
   ^

/ros_ws/src/repo-under-test/mavros/src/lib/uas_frame_conversions.cpp: In static member function ‘static mavros::UAS::Covariance6d mavros::UAS::transform_static_frame(const Covariance6d&, mavros::UAS::STATIC_TRANSFORM)’:
/ros_ws/src/repo-under-test/mavros/src/lib/uas_frame_conversions.cpp:133:3: error: ‘ROS_BREAK’ was not declared in this scope
   ROS_BREAK(false);
   ^

/ros_ws/src/repo-under-test/mavros/src/lib/uas_frame_conversions.cpp: In static member function ‘static mavros::UAS::Covariance6d mavros::UAS::transform_frame(const Covariance6d&, const Quaterniond&)’:
/ros_ws/src/repo-under-test/mavros/src/lib/uas_frame_conversions.cpp:165:2: error: ‘ROS_BREAK’ was not declared in this scope
  ROS_BREAK(false);
  ^

make[2]: *** [repo-under-test/mavros/CMakeFiles/mavros.dir/src/lib/uas_frame_conversions.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....

make[1]: *** [repo-under-test/mavros/CMakeFiles/mavros.dir/all] Error 2

make: *** [all] Error 2
```

gavanderhoorn, 2018-10-10:
I'm not entirely sure what is going on here.

If this is `ROS_BREAK(..)` from `rosconsole/ros/assert.h` (ie: [this one](https://github.com/ros/rosconsole/blob/b493395c60c3edb5261f2ae2f3f0d578f1641fc3/include/ros/assert.h#L116-L120)), then it does indeed not take any args. Usage patterns in `ros_comm` also suggest that no-args is the way to use it ([here](https://github.com/ros/ros_comm/search?q=ros_break&unscoped_q=ros_break)).

Looking at the history of the file ([here](https://github.com/ros/rosconsole/blame/60be2504d5341eb3bc80d86e5d16f1c53dc97ccd/tools/rosconsole/include/ros/assert.h), in the `ros_comm` repository, as `rosconsole` was split out of that some time ago), we see that at least for 8 years this macro has not taken any args.

gavanderhoorn, 2018-10-10:
Interestingly, this usage of `ROS_BREAK(..)` was introduced in mavlink/mavros@136f387f: lines go from `ROS_ASSERT(..)` (which does take 1 arg -- a condition) to `ROS_BREAK(..)`, which does not take any args.

This is all from mavlink/mavros#473.

It's fixed again in https://github.com/mavlink/mavros/pull/473/commits/5353982ea0f32783fcf3e296b25ab3b253550201.

gavanderhoorn, 2018-10-10:
(if not obvious: I've not read the bug description yet, so this could all be expected behaviour for this bug)

ChrisTimperley, 2018-10-23:
It looks like this is the expected behaviour. I'll build with the fix commit to confirm.

ChrisTimperley, 2018-10-23:
> It looks like this is the expected behaviour. I'll build with the fix commit to confirm.

The PUT builds successfully when the fix commit is applied.

closed 2018-10-23 by ChrisTimperley