robust-rosin/robust #130: PUT fails to build for mavros/84264f0

ChrisTimperley, 2018-10-10:
```
/ros_ws/src/repo-under-test/mavros/src/plugins/actuator_control.cpp: In member function ‘void mavplugin::ActuatorControlPlugin::set_actuator_control_target(uint64_t, uint8_t, const float*)’:
/ros_ws/src/repo-under-test/mavros/src/plugins/actuator_control.cpp:64:13: error: ‘mavlink_msg_set_actuator_control_target_pack_chan’ was not declared in this scope
     controls);
             ^

/ros_ws/src/repo-under-test/mavros/src/plugins/sys_status.cpp: In member function ‘void mavplugin::SystemStatusPlugin::autopilot_version_cb(const ros::TimerEvent&)’:
/ros_ws/src/repo-under-test/mavros/src/plugins/sys_status.cpp:616:26: error: ‘MAV_CMD_REQUEST_AUTOPILOT_CAPABILITIES’ was not declared in this scope
    cmd.request.command = MAV_CMD_REQUEST_AUTOPILOT_CAPABILITIES;
                          ^

make[2]: *** [repo-under-test/mavros/CMakeFiles/mavros_plugins.dir/src/plugins/actuator_control.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....

make[2]: *** [repo-under-test/mavros/CMakeFiles/mavros_plugins.dir/src/plugins/sys_status.cpp.o] Error 1

Linking CXX executable /ros_ws/devel/lib/mavros/mavros_node
[100%]
Built target mavros_node
make[1]: *** [repo-under-test/mavros/CMakeFiles/mavros_plugins.dir/all] Error 2

make: *** [all] Error 2
```

gavanderhoorn, 2018-10-10:
Link to bug: [mavros/84264f0.bug](https://github.com/robust-rosin/robust/blob/master/mavros/84264f0.bug).

ChrisTimperley, 2018-10-23:
Possibly related: https://gitter.im/mavlink/mavros/archives/2015/03/01

ChrisTimperley, 2018-10-30:
`mavlink_msg_set_actuator_control_target_pack_chan` should be provided by `mavlink_msg_set_actuator_control_target.h`, which appears to be missing:

```
root@e0c7ce9db154:/ros_ws# ls /opt/ros/indigo/include/mavlink/v1.0/common/
common.h                                               mavlink_msg_mission_item_reached.h
mavlink.h                                              mavlink_msg_mission_request.h
mavlink_msg_attitude.h                                 mavlink_msg_mission_request_list.h
mavlink_msg_attitude_quaternion.h                      mavlink_msg_mission_request_partial_list.h
mavlink_msg_attitude_quaternion_cov.h                  mavlink_msg_mission_set_current.h
mavlink_msg_attitude_target.h                          mavlink_msg_mission_write_partial_list.h
mavlink_msg_auth_key.h                                 mavlink_msg_named_value_float.h
mavlink_msg_autopilot_version.h                        mavlink_msg_named_value_int.h
mavlink_msg_battery_status.h                           mavlink_msg_nav_controller_output.h
mavlink_msg_change_operator_control.h                  mavlink_msg_optical_flow.h
mavlink_msg_change_operator_control_ack.h              mavlink_msg_optical_flow_rad.h
mavlink_msg_command_ack.h                              mavlink_msg_param_map_rc.h
mavlink_msg_command_int.h                              mavlink_msg_param_request_list.h
mavlink_msg_command_long.h                             mavlink_msg_param_request_read.h
mavlink_msg_data_stream.h                              mavlink_msg_param_set.h
mavlink_msg_data_transmission_handshake.h              mavlink_msg_param_value.h
mavlink_msg_debug.h                                    mavlink_msg_ping.h
mavlink_msg_debug_vect.h                               mavlink_msg_position_target_global_int.h
mavlink_msg_distance_sensor.h                          mavlink_msg_position_target_local_ned.h
mavlink_msg_encapsulated_data.h                        mavlink_msg_power_status.h
mavlink_msg_file_transfer_protocol.h                   mavlink_msg_radio_status.h
mavlink_msg_global_position_int.h                      mavlink_msg_raw_imu.h
mavlink_msg_global_position_int_cov.h                  mavlink_msg_raw_pressure.h
mavlink_msg_global_vision_position_estimate.h          mavlink_msg_rc_channels.h
mavlink_msg_gps2_raw.h                                 mavlink_msg_rc_channels_override.h
mavlink_msg_gps2_rtk.h                                 mavlink_msg_rc_channels_raw.h
mavlink_msg_gps_global_origin.h                        mavlink_msg_rc_channels_scaled.h
mavlink_msg_gps_inject_data.h                          mavlink_msg_request_data_stream.h
mavlink_msg_gps_raw_int.h                              mavlink_msg_safety_allowed_area.h
mavlink_msg_gps_rtk.h                                  mavlink_msg_safety_set_allowed_area.h
mavlink_msg_gps_status.h                               mavlink_msg_scaled_imu.h
mavlink_msg_heartbeat.h                                mavlink_msg_scaled_imu2.h
mavlink_msg_highres_imu.h                              mavlink_msg_scaled_pressure.h
mavlink_msg_hil_controls.h                             mavlink_msg_scaled_pressure2.h
mavlink_msg_hil_gps.h                                  mavlink_msg_serial_control.h
mavlink_msg_hil_optical_flow.h                         mavlink_msg_servo_output_raw.h
mavlink_msg_hil_rc_inputs_raw.h                        mavlink_msg_set_attitude_target.h
mavlink_msg_hil_sensor.h                               mavlink_msg_set_gps_global_origin.h
mavlink_msg_hil_state.h                                mavlink_msg_set_mode.h
mavlink_msg_hil_state_quaternion.h                     mavlink_msg_set_position_target_global_int.h
mavlink_msg_local_position_ned.h                       mavlink_msg_set_position_target_local_ned.h
mavlink_msg_local_position_ned_cov.h                   mavlink_msg_sim_state.h
mavlink_msg_local_position_ned_system_global_offset.h  mavlink_msg_statustext.h
mavlink_msg_log_data.h                                 mavlink_msg_sys_status.h
mavlink_msg_log_entry.h                                mavlink_msg_system_time.h
mavlink_msg_log_erase.h                                mavlink_msg_terrain_check.h
mavlink_msg_log_request_data.h                         mavlink_msg_terrain_data.h
mavlink_msg_log_request_end.h                          mavlink_msg_terrain_report.h
mavlink_msg_log_request_list.h                         mavlink_msg_terrain_request.h
mavlink_msg_manual_control.h                           mavlink_msg_timesync.h
mavlink_msg_manual_setpoint.h                          mavlink_msg_v2_extension.h
mavlink_msg_memory_vect.h                              mavlink_msg_vfr_hud.h
mavlink_msg_mission_ack.h                              mavlink_msg_vicon_position_estimate.h
mavlink_msg_mission_clear_all.h                        mavlink_msg_vision_position_estimate.h
mavlink_msg_mission_count.h                            mavlink_msg_vision_speed_estimate.h
mavlink_msg_mission_current.h                          testsuite.h
mavlink_msg_mission_item.h                             version.h
mavlink_msg_mission_item_int.h
```

ChrisTimperley, 2018-10-30:
Interestingly, `SET_ACTUATOR_CONTROL_TARGET` appears to be missing from `message_definitions/v1.0/common.xml` for the associated `mavlink` release: https://github.com/mavlink/mavlink-gbp-release/releases/tag/release%2Findigo%2Fmavlink%2F2015.2.2-0

ChrisTimperley, 2018-10-30:
@gavanderhoorn this is an interesting case. `SET_ACTUATOR_CONTROL_TARGET ` appears in the [next release of mavlink](https://github.com/mavlink/mavlink-gbp-release/releases/tag/release%2Findigo%2Fmavlink%2F2015.2.25-0), which was released after the issue was reported but before the fix was submitted, on 25th Feb 2015.

ChrisTimperley, 2018-10-30:
This builds as expected when the following `mavlink` release is used: https://github.com/mavlink/mavlink-gbp-release/releases/tag/release%2Findigo%2Fmavlink%2F2015.2.25-0

I'll update the `datetime` in the bug manifest file so that the manual fixes to the `deps.rosinstall` file won't be lost if the file is ever regenerated.

gavanderhoorn, 2018-10-30:
Must say these `mavros` bugs are proving to be quite a bit of work ..

commit ba126b49 (2018-10-30): 84264f0 now builds (fixes #130)
closed 2018-10-30 by ChrisTimperley