robust-rosin/robust #136: PUT fails to build for mavros/c172409

ChrisTimperley, 2018-10-10:
```
In file included from /ros_ws/src/repo-under-test/libmavconn/include/mavconn/interface.h:37:0,
                 from /ros_ws/src/repo-under-test/libmavconn/include/mavconn/serial.h:22,
                 from /ros_ws/src/repo-under-test/libmavconn/src/serial.cpp:22:
/ros_ws/src/repo-under-test/libmavconn/include/mavconn/mavlink_dialect.h:23:42: fatal error: mavlink/v2.0/common/common.hpp: No such file or directory
 #include <mavlink/v2.0/common/common.hpp>
                                          ^
compilation terminated.

In file included from /ros_ws/src/repo-under-test/libmavconn/include/mavconn/interface.h:37:0,
                 from /ros_ws/src/repo-under-test/libmavconn/include/mavconn/udp.h:22,
                 from /ros_ws/src/repo-under-test/libmavconn/src/udp.cpp:22:
/ros_ws/src/repo-under-test/libmavconn/include/mavconn/mavlink_dialect.h:23:42: fatal error: mavlink/v2.0/common/common.hpp: No such file or directory
 #include <mavlink/v2.0/common/common.hpp>
                                          ^
compilation terminated.

In file included from /ros_ws/src/repo-under-test/libmavconn/include/mavconn/interface.h:37:0,
                 from /ros_ws/src/repo-under-test/libmavconn/include/mavconn/tcp.h:24,
                 from /ros_ws/src/repo-under-test/libmavconn/src/tcp.cpp:22:
/ros_ws/src/repo-under-test/libmavconn/include/mavconn/mavlink_dialect.h:23:42: fatal error: mavlink/v2.0/common/common.hpp: No such file or directory
 #include <mavlink/v2.0/common/common.hpp>
                                          ^
compilation terminated.

make[2]: *** [repo-under-test/libmavconn/CMakeFiles/mavconn.dir/src/mavlink_helpers.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....

make[2]: *** [repo-under-test/libmavconn/CMakeFiles/mavconn.dir/src/interface.cpp.o] Error 1

[ 81%]
[ 81%]
Generating Python msg __init__.py for mavros_msgs
Generating Python srv __init__.py for mavros_msgs
[ 81%]
Built target mavros_msgs_generate_messages_py
[ 81%]
Built target mavros_msgs_generate_messages_cpp
Scanning dependencies of target mavros_msgs_generate_messages
[ 81%]
Built target mavros_msgs_generate_messages
make[2]: *** [repo-under-test/libmavconn/CMakeFiles/mavconn.dir/src/udp.cpp.o] Error 1

make[2]: *** [repo-under-test/libmavconn/CMakeFiles/mavconn.dir/src/tcp.cpp.o] Error 1

make[2]: *** [repo-under-test/libmavconn/CMakeFiles/mavconn.dir/src/serial.cpp.o] Error 1

make[1]: *** [repo-under-test/libmavconn/CMakeFiles/mavconn.dir/all] Error 2

make: *** [all] Error 2
```

ChrisTimperley, 2018-10-21:
Multiple versions of `mavros` attempt to import `mavlink/v2.0/common/common.hpp`. If we look at `mavlink/v2.0/common`, we see that there are no `*.hpp` files and that there is a `common.h` file:

```
root@476334f7337b:/opt/ros/indigo/include/mavlink/v2.0/common# ls
common.h                                               mavlink_msg_mission_item.h
mavlink.h                                              mavlink_msg_mission_item_int.h
mavlink_msg_actuator_control_target.h                  mavlink_msg_mission_item_reached.h
mavlink_msg_adsb_vehicle.h                             mavlink_msg_mission_request.h
mavlink_msg_altitude.h                                 mavlink_msg_mission_request_int.h
mavlink_msg_att_pos_mocap.h                            mavlink_msg_mission_request_list.h
mavlink_msg_attitude.h                                 mavlink_msg_mission_request_partial_list.h
mavlink_msg_attitude_quaternion.h                      mavlink_msg_mission_set_current.h
mavlink_msg_attitude_quaternion_cov.h                  mavlink_msg_mission_write_partial_list.h
mavlink_msg_attitude_target.h                          mavlink_msg_named_value_float.h
mavlink_msg_auth_key.h                                 mavlink_msg_named_value_int.h
mavlink_msg_autopilot_version.h                        mavlink_msg_nav_controller_output.h
mavlink_msg_battery_status.h                           mavlink_msg_optical_flow.h
mavlink_msg_camera_trigger.h                           mavlink_msg_optical_flow_rad.h
mavlink_msg_change_operator_control.h                  mavlink_msg_param_map_rc.h
mavlink_msg_change_operator_control_ack.h              mavlink_msg_param_request_list.h
mavlink_msg_command_ack.h                              mavlink_msg_param_request_read.h
mavlink_msg_command_int.h                              mavlink_msg_param_set.h
mavlink_msg_command_long.h                             mavlink_msg_param_value.h
mavlink_msg_control_system_state.h                     mavlink_msg_ping.h
mavlink_msg_data_stream.h                              mavlink_msg_position_target_global_int.h
mavlink_msg_data_transmission_handshake.h              mavlink_msg_position_target_local_ned.h
mavlink_msg_debug.h                                    mavlink_msg_power_status.h
mavlink_msg_debug_vect.h                               mavlink_msg_radio_status.h
mavlink_msg_distance_sensor.h                          mavlink_msg_raw_imu.h
mavlink_msg_encapsulated_data.h                        mavlink_msg_raw_pressure.h
mavlink_msg_estimator_status.h                         mavlink_msg_rc_channels.h
mavlink_msg_extended_sys_state.h                       mavlink_msg_rc_channels_override.h
mavlink_msg_file_transfer_protocol.h                   mavlink_msg_rc_channels_raw.h
mavlink_msg_follow_target.h                            mavlink_msg_rc_channels_scaled.h
mavlink_msg_global_position_int.h                      mavlink_msg_request_data_stream.h
mavlink_msg_global_position_int_cov.h                  mavlink_msg_resource_request.h
mavlink_msg_global_vision_position_estimate.h          mavlink_msg_safety_allowed_area.h
mavlink_msg_gps2_raw.h                                 mavlink_msg_safety_set_allowed_area.h
mavlink_msg_gps2_rtk.h                                 mavlink_msg_scaled_imu.h
mavlink_msg_gps_global_origin.h                        mavlink_msg_scaled_imu2.h
mavlink_msg_gps_inject_data.h                          mavlink_msg_scaled_imu3.h
mavlink_msg_gps_raw_int.h                              mavlink_msg_scaled_pressure.h
mavlink_msg_gps_rtcm_data.h                            mavlink_msg_scaled_pressure2.h
mavlink_msg_gps_rtk.h                                  mavlink_msg_scaled_pressure3.h
mavlink_msg_gps_status.h                               mavlink_msg_serial_control.h
mavlink_msg_heartbeat.h                                mavlink_msg_servo_output_raw.h
mavlink_msg_highres_imu.h                              mavlink_msg_set_actuator_control_target.h
mavlink_msg_hil_controls.h                             mavlink_msg_set_attitude_target.h
mavlink_msg_hil_gps.h                                  mavlink_msg_set_gps_global_origin.h
mavlink_msg_hil_optical_flow.h                         mavlink_msg_set_home_position.h
mavlink_msg_hil_rc_inputs_raw.h                        mavlink_msg_set_mode.h
mavlink_msg_hil_sensor.h                               mavlink_msg_set_position_target_global_int.h
mavlink_msg_hil_state.h                                mavlink_msg_set_position_target_local_ned.h
mavlink_msg_hil_state_quaternion.h                     mavlink_msg_setup_signing.h
mavlink_msg_home_position.h                            mavlink_msg_sim_state.h
mavlink_msg_landing_target.h                           mavlink_msg_statustext.h
mavlink_msg_local_position_ned.h                       mavlink_msg_sys_status.h
mavlink_msg_local_position_ned_cov.h                   mavlink_msg_system_time.h
mavlink_msg_local_position_ned_system_global_offset.h  mavlink_msg_terrain_check.h
mavlink_msg_log_data.h                                 mavlink_msg_terrain_data.h
mavlink_msg_log_entry.h                                mavlink_msg_terrain_report.h
mavlink_msg_log_erase.h                                mavlink_msg_terrain_request.h
mavlink_msg_log_request_data.h                         mavlink_msg_timesync.h
mavlink_msg_log_request_end.h                          mavlink_msg_v2_extension.h
mavlink_msg_log_request_list.h                         mavlink_msg_vfr_hud.h
[... cut ...]
mavlink_msg_manual_setpoint.h                          mavlink_msg_vicon_position_estimate.h
mavlink_msg_memory_vect.h                              mavlink_msg_vision_position_estimate.h
mavlink_msg_message_interval.h                         mavlink_msg_vision_speed_estimate.h
mavlink_msg_mission_ack.h                              mavlink_msg_wind_cov.h
mavlink_msg_mission_clear_all.h                        testsuite.h
mavlink_msg_mission_count.h                            version.h
mavlink_msg_mission_current.h
```

ChrisTimperley, 2018-10-23:
The file with a bad include is [autogenerated](https://github.com/mavlink/mavros/blob/1ed30ca7a19a697a24c2c01426ca3d9fe13082a8/libmavconn/include/mavconn/mavlink_dialect.h.em) within the `libmavconn` directory of the PUT. And here's the line that's causing the build failure:
```
@[for dialect in MAVLINK_V20_DIALECTS]#include <mavlink/v2.0/@(dialect)/@(dialect).hpp>
@[end for]
```

ChrisTimperley, 2018-10-23:
The "missing" files are automatically generated as part of the `mavlink` package: https://github.com/mavlink/mavlink-gbp-release. For this particular bug, we use the following release: https://github.com/mavlink/mavlink-gbp-release/releases/tag/release%2Findigo%2Fmavlink%2F2016.5.20-0. All of the automatically generated header files end in `.h` (not `.hpp`).

Interestingly, it looks like the [previous release](https://github.com/mavlink/mavlink-gbp-release/releases/tag/release%2Findigo%2Fmavlink%2F2016.5.15-0) of `mavlink` doesn't generate `v2.0` of the `mavlink` protocol, so it would seem that going backwards in time is not a solution here.

ChrisTimperley, 2018-10-23:
This issue completely disappears the ROS distro is switched from Indigo to Kinetic.

closed 2018-10-23 by ChrisTimperley

gavanderhoorn, 2018-10-23:
Wait.

We cannot just switch the ros distro to "fix" this.

ChrisTimperley, 2018-10-23:
> Wait.
> 
> We cannot just switch the ros distro to "fix" this.
> 
> If the bug was reported for a particular ROS version, it makes little sense to me to switch to a different one (especially not Indigo -> Kinetic) as that also means switching OS (Trusty -> Xenial).

The bug wasn't reported for a particular ROS distro: https://github.com/mavlink/mavros/issues/609

ChrisTimperley, 2018-10-23:
Also

> > Interestingly, it looks like the [previous release](https://github.com/mavlink/mavlink-gbp-release/releases/tag/release%2Findigo%2Fmavlink%2F2016.5.15-0) of `mavlink` doesn't generate `v2.0` of the `mavlink` protocol, so it would seem that going backwards in time is not a solution here.
> 
> I'm not sure I follow here: it seems like using a newer version of a dependency actually helps with reproducing an older bug.
> 
> That sounds a bit strange to me.
> 
> Are we sure we are using the correct rosdistro and rosdep DBs?

The bug was reported on 2016-09-30. The version of `mavlink` selected by the time machine was released on 2016-05-15. That particular release, of `mavlink` so far as I can tell, is the last release that was made for `indigo`. Going backwards to a previous release does not fix the problem. The bug does not rely on Indigo. Trying to get this bug to reproduce on Indigo is a waste of time.

commit 6df1dcf7 (2018-10-23): switched mavros/c172409 to kinetic (fixes #136)